dsmrreader/dsmr-reader #305: Trend staafdiagrammen afgelopen week / afgelopen maand altijd gelijk

dbmnl, 2017-04-29:
Het viel mij al een tijdje op dat op de pagina trends de staafdiagrammen altijd exact gelijk zijn. Ik denk dat de berekening voor beide gelijk zijn aan elkaar. Kan jij daar eens naar kijken? Tnx

[image]

dennissiemensma, 2017-04-29:
Bedankt voor het melden. Ik heb even gekeken en er lijkt inderdaad een foutje in te zitten. De gegevens worden wel goed opgehaald, maar om de een of andere manier gaat het mis bij het tekenen van de grafieken.

Ik zal hier later een fix voor maken die dit oplost.

Varsame, 2017-07-05:
Hi Dennis, 
Heeft deze bug nog je aandacht? :)

dennissiemensma, 2017-07-05:
Ik ben als eerste bezig met #9, een langekoesterde wens. Daarna doe ik zeer waarschijnlijk een release met fixes voor de frontend/GUI.

dennissiemensma, 2017-10-08:
Ik heb de bug gevonden en opgelost voor de volgende release. Nogmaals bedankt voor het melden.

closed 2017-10-08 by dennissiemensma

dbmnl, 2017-10-09:
Wederom bedankt Dennis. 
Ik maak hier graag gebruik van en wacht dus met smart op 1.10.