dsmrreader/dsmr-reader #306: Can't see the application

HFeenstra, 2017-04-29:
Hi all,

What a tremandous work you have done. I could do all the steps, everything returned exactly like you guys stated; but,..... When I go to my ip address of the Pi I do get this:

Welcome to nginx on Debian!

If you see this page, the nginx web server is successfully installed and working on Debian. Further configuration is required.

For online documentation and support please refer to nginx.org

Please use the reportbug tool to report bugs in the nginx package with Debian. However, check existing bug reports before reporting a new bug.

Thank you for using debian and nginx.

What have I done wrong, again I followed every step, but can't see the application.... Can you please tell me what I have to do to get it working?

Greetings,

Hette Feenstra
The Netherlands

dennissiemensma, 2017-04-29:
Well the good news is that the web server seems to run fine. I think the **default vhost** is still active.

Can you check whether it's still there, and if so, delete it and reload Nginx?

* `sudo ls -l /etc/nginx/sites-enabled/`
* `sudo rm /etc/nginx/sites-enabled/default`
* `sudo service nginx configtest`
* `sudo service nginx reload`

For your reference, this is what I see when running the first command:
```
sudo ls -l /etc/nginx/sites-enabled/

total 4
-rw-r--r-- 1 root root 702 dec 20 20:46 dsmr-webinterface
```

HFeenstra, 2017-04-29:
Dennis (super!) (mag in Nederlands, maar laten we maar Engels houden)
first command:
sudo ls -l /etc/nginx/sites-enabled/
total 4
-rw-r--r-- 1 root root 790 Apr 29 17:01 dsmr-webinterface

second command:
sudo rm /etc/nginx/sites-enabled/default
rm: cannot remove ‘/etc/nginx/sites-enabled/default’: No such file or directory

third command:
sudo service nginx configtest
[ ok ] Testing nginx configuration:.

fourth command:
sudo service nginx reload
(nothing returned, so I guess ok?)

Afterwards, reloaded site and indeed all is up!
Thanks will dive in it to set up the rest!.

Dennis thanks also for your quick reply!

dennissiemensma, 2017-04-29:
Goed om te horen, zowel Nederlands als Engels is prima.

closed 2017-04-29 by dennissiemensma