dsmrreader/dsmr-reader #307: Upgrade 1.6.1 naar 1.6.2

TopdRob, 2017-04-30:
Bij de upgrade van 1.6.1 naar 1.6.2 heeft het upgrade proces een bestand /etc/nginx/sites-enabled/dsmr-webinterface.save aangemaakt

Maar hierdoor start mijn nginx niet meer op omdat hij een duplicaat heeft.

Na het verwijderen van het .save bestand werkt het weer.

dennissiemensma, 2017-04-30:
Bedankt voor je melding. Weet je nog precies hoe je hebt geupgrade? 

Als het goed is doet het upgrade-script namelijk niets met vhosts of Nginx zelf. Kan het zijn dat je iets gekopieerd of gereload hebt?

TopdRob, 2017-04-30:
Met ./deploy.sh via sudo su dsmr

dennissiemensma, 2017-04-30:
Ik heb nog even voor je gekeken in de code en scripts, maar ik zie geen mogelijkheid dat het deploy-script iets met de vhosts doet. Voor nu kan ik hier dus niets aan doen, maar laat het vooral weten als je het een volgende keer wéér ervaart. :]

closed 2017-04-30 by dennissiemensma