dsmrreader/dsmr-reader #309: changing port for DSMR web interface

HFeenstra, 2017-04-30:
Hello,

I am completly new to this reader, but I love IT!! Thanks for far.

Are there somewhere steps, a script that can help me change the port. I am thinking of ussing my RPi also for Pi-hole, but since it is using the same port (80 I thought), it will not work well.

I hope you can let me know whether this is possible?

Keep up the very good and nice work.

dennissiemensma, 2017-04-30:
Hi, this certainly is possible, as it doesn't matter for dmsrreader or nginx what port it's running on. 

Someone else asked the exact same question a while ago, so I'll direct you toward that topic. You [can find it here #169](https://github.com/dennissiemensma/dsmr-reader/issues/169#issuecomment-263716754).

Do not hesistate if you still have any questions.

closed 2017-04-30 by dennissiemensma

HFeenstra, 2017-05-26:
Left comment in the #169 thread