dsmrreader/dsmr-reader #310: Nog een paar vragen

HFeenstra, 2017-05-02:
Hoi,

Ben ik weer, misschien heb je nog wat aanwijzingen voor mij?
- Ik heb de DSMR software op een aparte Pi draaien, maar wil graag de output uitlezen op een andere Pi waar ik HomeAssistant heb draaien. Wil graag gebruik maken van deze component: https://home-assistant.io/components/sensor.dsmr/ Kan dit? En wat zijn dan de stappen om het te laten werken?
- Ik heb 3 fasen stroom, is er een mogelijkheid dat de grafiek van 3 fasen, altijd wordt getoond; op het moment dat je hebt ingesteld dat je 3 fasen grafiek ook wilt zien? Nu moet ik elke keer opnieuw eerst klikken voor ik de grafiek daadwerkelijk ook kan zien.

Succes verder.

WatskeBart, 2017-05-02:
Om de output op een andere Pi uit te kunnen lezen, zou je de ser2net methode kunnen proberen.

Op je Pi met dsmr-reader installeer je ser2net:
`sudo apt-get update && apt-get install ser2net`

Daarna ser2net starten met een one-liner om te testen:
`ser2net -C 2001:raw:600:/dev/ttyUSB0:115200:8DATABITS:NONE:1STOPBIT`
_Om dit proces te stoppen geef je het commando:_
`pkill ser2net`

Op je andere Pi met HomeAssistant (waarschijnlijk Hasbbian) geef je de juiste waarden (verander het IP adres naar het adres van jouw Pi welke dsmr-reader draait) in je configuration.yaml, zoals beschreven op de door [jou aangegeven website](https://home-assistant.io/components/sensor.dsmr/).

Wanneer dit werkt, kan je de standaard **/etc/ser2net.conf** aanpassen en dan hoef je alleen maar het commando: `ser2net` te geven en dan start hij automatisch met de door jou aangepaste configuratie bestand.

dennissiemensma, 2017-05-02:
@HFeenstra wat betreft je vraag over het uitlezen op een aparte Pi, hoe bedoel je dat exact? Je zou bovenstaande tip van Watkebart kunnen proberen. Anderzijds is wellicht e.e.a. mogelijk via de (nieuwe) API van dsmrreader. Alleen zul je daarvoor wat moeten scripten.

Je andere vraag over het altijd tonen van de 3 fasen: Opzich kan ik daar wel wat voor maken. Alleen is die nu bewust standaard verborgen omdat men in de praktijk alleen de eerste dagen of weken kijkt naar de fasen en daarna veel minder (of helemaal niet meer). Wellicht kun je het even aankijken en mocht je alsnog na een tijd dagelijks kijken naar de fasen, dan kan ik er in de toekomst wel iets voor maken.

HFeenstra, 2017-05-03:
Hallo,

Dank voor de stappen die je mee hebt gestuurd.
Ik heb nog sudo !! geprobeerd bij apt-get install, echter ik krijg steeds:
*E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission
denied)*
*E: Unable to lock the administration directory (/var/lib/dpkg/), are you
root?*

Wat doe ik hier fout en wat moet ik doen om het werkend te krijgen? Moet ik
mogelijk eerst in de venv van DSMR gaan om dit te kunnen laten draaien?

Groet,

H. Feenstra

dennissiemensma, 2017-05-03:
Je kunt proberen om bovenstaande commando's uit te voeren als `root`, wanneer `sudo` niet werkt. Meestal kun je `root` worden vanaf de standaard `pi` gebruiker met `sudo -s` or `sudo su`.

closed 2017-09-11 by dennissiemensma