dsmrreader/dsmr-reader #313: API optie via webbrowser

Jeltel, 2017-05-05:
Top werk met de API, die werkt vast goed via een andere applicatie (zoals het voorstelde Python en curl). Maar om de API snel te leren kennen gebruik ik regelmatig even een browser en speel wat met de url en variabelen. Al naar gelang de uitkomst pas ik de url aan en ga ik deze implementeren in python/curl/winautomation etc.

Maar het lijkt erop alsof de authkey alleen via een header meegegeven kan worden. Is het mogelijk om deze ook in de URL bij te voegen? ala:

[](http://192.168.x.x/api/v2/datalogger/dsmrreading?X-AUTHKEY=helelangekey)

dennissiemensma, 2017-05-07:
Bedankt voor je suggestie. Doorgaans is het meegeven van API keys via de URL iets dat afgeraden wordt. Echter, gezien de applicatie lokaal draait is het risico al een stuk minder, dus ik zal het later overwegen. Of ik bouw iets in zodat je de API via een webinterface of iets kan uitproberen.

dennissiemensma, 2017-09-29:
Ik heb hier nog even naar gekeken, maar het is relatief veel werk om dit goed werkend te krijgen. Duus ik laat dit voor wat het is.

Overigens heb ik nog wel een zeer goed alternatief voor je dat je ook via je browser (sowieso Chrome/Firefox kan gebruiken: [Postman](https://www.getpostman.com/)
- Firefox: https://addons.mozilla.org/nl/firefox/tag/postman
- Chrome: https://chrome.google.com/webstore/detail/postman/fhbjgbiflinjbdggehcddcbncdddomop

Hiermee kun je makkelijk API calls maken en tevens eventuele headers instellen die nodig zijn voor autheticatie, zoals hier. Ik heb hier zelf ook veelvuldig gebruik van gemaakt. 
Stom van me dat ik daar, op het initiele moment van je ticket beantwoorden, niet aan gedacht heb. Hopelijk helpt je dit alsnog.

closed 2017-09-29 by dennissiemensma