dsmrreader/dsmr-reader #315: Support for new meter/moving to a new home

balk77, 2017-05-06:
As the project matures, more and more users will or receive a new meter or move to a new home. From​ a DSMR-reader perspective moving is like receiving a new meter with a non zero reading.

Can you please add support for this? Thanks

dennissiemensma, 2017-05-07:
Thanks for the suggestion. There is already a ticket for this: #225. As there currently is a workaround as well for those having their meter replaced (or move), I will surely implement this in the future, but not anytime soon.

dennissiemensma, 2017-05-07:
Closed due duplicate, see #225

closed 2017-05-07 by dennissiemensma

BBasss, 2018-02-02:
Hoi Dennis,

In de omgeving Amersfoort gaat Stedin nu beginnen met de gasmeterwissels ivm. onveilige meters ( Zelf spreken ze van het "niet voldoet aan onze hoge kwaliteitseisen"....... jaja
Ik weet niet hoeveel blije DSMR reader gebruikers zijn, maar eentje in ieder geval ;-)

Kan ik nog behulpzaam zijn met loggings oid?

dennissiemensma, 2018-02-02:
Bedankt voor je aankondiging! Ik ga er vanuit dat hierdoor de techniek van de slimme meters zelf (en de telegrammen) niet wijzigt. 
Het kan wel zijn dat het verbruik op de eerste dag van/na de wisseling van gasmeters niet klopt, omdat de applicatie nog geen ondersteuning voor meerdere meters heeft. Je kunt dat eventueel zelf eenmalig bijwerken in de admin-interface (dagverbruik).

Mocht je desondanks zien dat er helemaal geen gasverbruik meer gemeten wordt, dan hoor ik het uiteraard graag!