dsmrreader/dsmr-reader #316: There are too many unprocessed telegrams after update

darthf1, 2017-05-07:
Hi. Ik heb afgelopen week een update uitgevoerd van DSMR reader via ./deploy.sh en sindsdien worden geen berichten meer verwerkt. Ik heb de volgende aantallen:

> There are too many unprocessed telegrams.
> Unprocessed telegrams: 37,128

> Data processing is lagging behind: 4 days, 7 hours ago

Hoe kan ik dit proces weer op gang brengen?

dennissiemensma, 2017-05-07:
Bedankt voor het melden van dit probleem. Het lijkt erop dat je niet in de 'virtualenv' zit waarin Django geinstalleerd is. 
Kun je eens kijken of je [de stappen halverwege hoofdstuk 6](http://dsmr-reader.readthedocs.io/nl/latest/installation.html#virtualenv) hebt uitgevoerd? Voor het upgraden zul je namelijk zowel de `dsmr` gebruiker moeten zijn als de virtualenv activeren. Het toevoegen van de commands in je .bashrc zou dat automatisch moeten doen wanneer je `sudo su - dsmr` doet.

Zou je ook willen proberen of een herstart van het dataverwerkingproces dit verhelpt? Het kan wel enige tijd duren voordat die weer bij is, maar de achterstand loopt doorgaans in. Dit mag als root/sudo user uitgevoerd worden:

```
sudo supervisorctl restart dsmr_backend
```

darthf1, 2017-05-07:
Die stappen in hoofdstuk 6 heb ik uitgevoerd tijdens de installatie. Als ik je goed begrijp zou de virtualenv automatisch moeten worden geactiveerd? De upgrade heb ik als dsmr user uitgevoerd middels `sudo su - dsmr`

Ik heb `sudo supervisorctl restart dsmr_backend` uitgevoerd en het aantal unprocessed telegrams loopt gestaag omlaag! Dus dat lijkt nu goed te gaan.

Is er iets niet goed gegaan tijdens het update proces? Heb ik een stap overgeslagen?

dennissiemensma, 2017-05-07:
Dat is voor mij moeilijk te zeggen op afstand. Je kunt controleren of het automatisch activeren van de virtualenv goed werkt met:

`sudo su - dsmr`

Als het goed is staat er nu iets in je terminal als: `(dsmrreader)dsmr@***:~/dsmr-reader$ `
Wanneer het *niet* begint met `(dsmrreader)` is die waarschijnlijk *niet* geactiveerd.

Verder zou het Python-pad bij het activeren van de virtualenv moeten verwijzen naar die van de virtualenv:
`which python3` -> `/home/dsmr/.virtualenvs/dsmrreader/bin/python3`

Django moet geinstalleerd zijn:
`pip freeze | grep -i django`

Mocht je tegen nog meer zaken aanlopen, hoor ik het graag. Voor nu ga ik er vanuit dat de verwerkingsachterstand ingehaald wordt.

closed 2017-05-07 by dennissiemensma

darthf1, 2017-05-07:
Alle zaken die je opnoemt gaan goed, dus daar lijkt het toch neit aan te liggen. Maar mijn probleem is inderdaad opgelost, de achterstand is bijna helemaal weggewerkt. Bedankt voor je reacties op deze zondag!