dsmrreader/dsmr-reader #318: Python module six is missing from requirements

xfade, 2017-05-11:
On a clean install in a virtual env, migrate failed due to missing python module six. Adding it to the requirements fixed it.

dennissiemensma, 2017-05-11:
Thank you for reporting this issue. Could you provide some more info about what command you've executed? 

As far as I can tell, the `six` module should be a builtin requirement for Django and therefor will be installed automatically along with Django. For that same reason, it's absent from the requirements file of this project.

I've tried to reproduce it, but I cannot seem to get the virtualenv in the same state:
```
  virtualenv ~/.virtualenvs/test --no-site-packages --python python3
  source ~/.virtualenvs/test/bin/activate
  pip3 install -r dsmrreader/provisioning/requirements/base.txt
  pip3 freeze | grep six
```

The last command results in: `six==1.10.0`

xfade, 2017-05-11:
Raspberry Pi 2 running Raspbian.

Looking at my history:

```
git clone https://github.com/dennissiemensma/dsmr-reader.git
mkdir ~/.virtualenvs
virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
nano .bashrc
<logout/login>
cp dsmrreader/provisioning/django/postgresql.py dsmrreader/settings.py
pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt
./manage.py migrate
```

dennissiemensma, 2017-05-12:
Thank you, I will try it sometime on my raspberry. 

I asume there is no rush, for now, since you've already managed to work around the issue yourself anyway.

dennissiemensma, 2017-09-11:
I have tried to reproduce it on my Pi as well, but it just seems to install {{six}} anyway. I'll close this for now, as there are no other reports of this happening. Still thank you for bringing this up.

closed 2017-09-11 by dennissiemensma