dsmrreader/dsmr-reader #319: Datum dashboard past niet automatisch aan

goegol, 2017-05-20:
Hoi,

Als je dashboard meerdere dagen laat open staan (of verwerking van telegrammen lopen achter, en worden weer ingehaald), dan blijft de datum op dashboard staan van vorige dag of (in mijn geval, omdat verwerking was gestopt) op paar dagen geleden.

grafiek en tijd update wel automatisch.
Klein bugje, mogelijk helemaal niet de moeite waard om te fixxen

Groeten,

Hans-Paul

dennissiemensma, 2017-05-21:
Bedankt voor het melden. Het is inderdaad 'by design', maar meer in de trend van dat ik er geen rekening mee heb gehouden dat iemand een pagina langer dan 5 minuten o.i.d. open laat staan.

Wellicht dat ik er bij het refactoren van de interface rekening mee kan houden.

dennissiemensma, 2017-09-11:
Mogelijk neem ik dit mee in de redesign van de applicatie. Sowieso is de datumweergave in het dashboard niet altijd correct (bij achterstand), dus ik mag daar een betere oplossing voor vinden. Voor nu sluit ik deze.

closed 2017-09-11 by dennissiemensma