dsmrreader/dsmr-reader #320: Ideetje: auto refresh op statussen op statuspagina?

goegol, 2017-05-20:
Doordat verwerking tijdje stil stond (reden onbekend, een ./reload.sh verhelpte het issue) liep ik 25.000 telegrammen achter.
Handmatig refreshen is old skool, mogelijk om die automatisch te laten refreshen?

dennissiemensma, 2017-05-21:
In principe kan ik alleen de oorzaak oplossen. Mocht je nog logs hebben met een foutmelding van de mogelijke oorzaak, dan zie ik die graag tegemoet.

Overigens kun je zelf ook het script periodiek uitvoeren via de crontab, maar dat is meer een pleister. Je kunt een script aanmaken in `/etc/cron.hourly/dsmrreader-reload` met:
```
#!/bin/sh
sudo -u dsmr sh /home/dsmr/dsmr-reader/reload.sh
```

Zorg dat `/etc/cron.hourly/dsmrreader-reload` op owner `root:root` staat met rechten `0755`. Testen kan met `run-parts -v /etc/cron.hourly/` (let op: alle andere scripts in die map worden ook uitgevoerd).

dennissiemensma, 2017-05-21:
Ik zie dat ik het verkeerd gelezen hebt, je doelt hier op de Status-pagina. Daar kan ik zeker wel wat mee doen.

dennissiemensma, 2017-09-11:
Ik heb een auto-refresh van 10 seconden toegevoegd voor de volgende release.

closed 2017-09-11 by dennissiemensma