dsmrreader/dsmr-reader #321: Applicatie heeft reload nodig na onderhoud mysql

goegol, 2017-05-20:
Even aan testen geweest waarom mijn telegrammen niet meer werden verwerkt:
Van de week onderhoud aan mysql gedaan en mysql gerestart, sindsdien liepen de onverwerkte telegrammen op. Zojuist getest en kan reproduceren.

Nadat mysql weer online is, haalt deze wel netjes de gegegevens op, maar verwerkt de telegrammen niet meer. Na reload heeft van de dsmrreader werkt alles weer naar behoren.

Misschien mogelijkheid tot verbetering? (bijvoorbeeld na zoveel onverwerkte telegrammen, retry op verwerking van de telegrammen?)

dennissiemensma, 2017-05-21:
Ik denk dat het een bugje is in Django. Voor nu zou je het reload script in #320 bijvoorbeeld elke dag kunnen laten uitvoeren als workaround (in `/etc/.cron.daily`). Ik zal sowieso kijken of ik er wat aan kan doen.

dennissiemensma, 2017-09-11:
Ik heb hier destijds kort naar gekeken, maar had geen goede oplossing hiervoor. In principe is het de taak van het onderliggende framework (Django) om dat goed op te vangen. Op termijn upgrade ik het project weer 3 versies omhoog, naar de nieuwste LTS versie van Django. Mogelijk lost dat een toekomstig gelijke situatie op.

Daarnaast is het denk ik een goede aanvulling voor #208 qua monitoring. Het niet verwerken van gegevens zou bij implementatie daarvan ook een melding kunnen geven.

closed 2017-09-11 by dennissiemensma