dsmrreader/dsmr-reader #323: dsmr services in loop (not starting)

dommels, 2017-05-22:
Hi,

I'm in the process of setting up dsmr-reader on my orange pi zero and so far I have managed to successfully go through all the steps (with some effort) .
Only now both dsmr_datalogger and dsmr_webinterface process seem to be stuck in a startup loop. 
I have attached logs. What could be causing this?

[log.txt](https://github.com/dennissiemensma/dsmr-reader/files/1019938/log.txt)

dennissiemensma, 2017-05-22:
Regarding the datalogger: did you check whether you could read your serial cable during installation with the `du` command? 
And what kind of smart meter do you have installed, with what DSMR-version?

As for the webinterface, can you try running directly the command supervisor is attempting to run:
```
sudo su - dsmr
/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
```

Does it state any other error?

dommels, 2017-05-22:
actually it wasn't connected yet. now the datalogger is running (Kaifa MA105c)

the webinterface still gives the same error

dennissiemensma, 2017-05-23:
Do you mean it gives the same error when the cable is connected, or you see the same error when you manually execute the command above?

dommels, 2017-05-23:
I'm getting a somewhat different error (datalogger still up and running now)

supervisor> tail -f dsmr_webinterface
==> Press Ctrl-C to exit <==
[2017-05-23 17:08:09 +0100] [6021] [ERROR] Can't connect to /var/tmp/gunicorn--dsmr_webinterface.socket
[2017-05-23 17:08:10 +0100] [6025] [INFO] Starting gunicorn 19.7.0
[2017-05-23 17:08:10 +0100] [6025] [ERROR] Retrying in 1 second.
[2017-05-23 17:08:11 +0100] [6025] [ERROR] Retrying in 1 second.
[2017-05-23 17:08:12 +0100] [6025] [ERROR] Retrying in 1 second.
[2017-05-23 17:08:14 +0100] [6025] [ERROR] Retrying in 1 second.
[2017-05-23 17:08:15 +0100] [6025] [ERROR] Retrying in 1 second.

(dsmrreader)dsmr@DietPi:~/dsmr-reader$
[2017-05-23 17:07:36 +0100] [6000] [INFO] Starting gunicorn 19.7.0
Error: Already running on PID 5997 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

dennissiemensma, 2017-05-23:
You might want to stop the Gunicorn process in Supervisor first, before triggering it manually (the 'stale' error). And then run Gunicorn with the command mentioned before.

Does this still trigger the 'stale' error?
```
sudo supervisorctl stop dsmr_webinterface

sudo su - dsmr

/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
```

dommels, 2017-05-23:
I stopped the process and now it seems to give the same error indeed:
(dsmrreader)dsmr@DietPi:~/dsmr-reader$ /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
[2017-05-23 18:07:05 +0100] [7988] [INFO] Starting gunicorn 19.7.0
[2017-05-23 18:07:05 +0100] [7988] [ERROR] Retrying in 1 second.
[2017-05-23 18:07:06 +0100] [7988] [ERROR] Retrying in 1 second.
[2017-05-23 18:07:07 +0100] [7988] [ERROR] Retrying in 1 second.
[2017-05-23 18:07:08 +0100] [7988] [ERROR] Retrying in 1 second.
[2017-05-23 18:07:09 +0100] [7988] [ERROR] Retrying in 1 second.
[2017-05-23 18:07:10 +0100] [7988] [ERROR] Can't connect to /var/tmp/gunicorn--dsmr_webinterface.socket

there seems to be no such file:
(dsmrreader)dsmr@DietPi:/var/tmp$ ls
dsmrreader--dsmr_backend.pid  dsmrreader--dsmr_datalogger.pid  gunicorn--dsmr_webinterface.pid

dennissiemensma, 2017-05-23:
Is it possible that the OS has some kind of secutiry layer like CentOS's `SElinux`? It seems related to access like this here: https://stackoverflow.com/questions/39504485/gunicorn-cant-connect-to-socket-error-running-in-vagrant

Can you try starting it in Supervisor again, and check whether `/var/tmp/gunicorn--dsmr_webinterface.socket` exists and what the owner/permissions are?

dennissiemensma, 2017-05-23:
This is how `/var/tmp/` looks for me:

```
-rw-r--r-- 1 dsmr dsmr    5 mei 21 07:46 dsmrreader--dsmr_backend.pid
-rw-r--r-- 1 dsmr dsmr    4 mei 21 07:46 gunicorn--dsmr_webinterface.pid
srwxrwxrwx 1 dsmr dsmr    0 mei  3 19:46 gunicorn--dsmr_webinterface.socket
```

dommels, 2017-05-23:
I tried but It is still missing:
-rw-r--r-- 1 dsmr dsmr 3 May 22 21:39 dsmrreader--dsmr_backend.pid
-rw-r--r-- 1 dsmr dsmr 3 May 22 21:39 dsmrreader--dsmr_datalogger.pid
-rw-r--r-- 1 dsmr dsmr 5 May 23 19:25 gunicorn--dsmr_webinterface.pid

dommels, 2017-05-23:
It's running! I eventually managed to get gunicorn service up and running but the issue was still there.
according to this topic there is some bug in gunicorn v19 (https://groups.google.com/forum/#!topic/obey-the-testing-goat-book/z2ql1Jw7heE) so I tried my luck and downgraded: pip install gunicorn==18 

(dsmrreader)dsmr@DietPi:~/dsmr-reader$ /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
2017-05-23 20:25:53 [2121] [INFO] Starting gunicorn 18.0
2017-05-23 20:25:53 [2121] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (2121)
2017-05-23 20:25:53 [2121] [INFO] Using worker: sync
2017-05-23 20:25:53 [2124] [INFO] Booting worker with pid: 2124
2017-05-23 20:26:25 [2121] [INFO] Handling signal: winch
2017-05-23 20:26:25 [2121] [INFO] SIGWINCH ignored. Not daemonized

thank you

dennissiemensma, 2017-05-23:
Can you try running Gunicorn with an extra ``--log-level debug`` parameter?

```
sudo supervisorctl stop dsmr_webinterface

sudo su - dsmr

/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid --log-level debug dsmrreader.wsgi
```

Does it provide more information regarding the error?

dennissiemensma, 2017-05-23:
Another suggestion you can try, is avoiding using sockets between Nginx and Gunicorn at all. Network stack is supported as well.

Edit `/etc/nginx/sites-enabled/dsmr-webinterface`:

```
# Comment this line by adding #
# server unix:///var/tmp/gunicorn--dsmr_webinterface.socket fail_timeout=0;

# Add this line under it:
server 127.0.0.1:10000 fail_timeout=0;
```

Test and reload Nginx:
```
service nginx configtest 

# [ ok ] Testing nginx configuration:.
service nginx reload
```

Now run it manually first, using this route again:

```
sudo supervisorctl stop dsmr_webinterface

sudo su - dsmr

# The --bind parameter differs here.
/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind 127.0.01:10000 --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
```

Does it run now?

dennissiemensma, 2017-05-23:
I'm sorry, I totally misread your last message that you had it running (facepalm to myself). If you need any other help, please let me know.

closed 2017-05-23 by dennissiemensma

dommels, 2017-05-23:
yes, everything seems to be running great now. also quite smooth considering the hardware. i'm quite impressed. thank you very much.

dennissiemensma, 2017-05-23:
Great to hear.!

Please note that your SD card (if you use one) still is the 'weakest' link, as it will degrade by the many writes of the database, so be sure to have the backup's configured as well. Or host the database somewhere else. See the documentation/FAQ as well. :]