dsmrreader/dsmr-reader #325: Update command geeft fout aan

HFeenstra, 2017-05-26:
Ok, ik heb de volgende stappen uitgevoerd voor het updaten:
eerst: sudo su - dsmr
door het command in de virtualenv gekomen; daarna ./deploy.sh uitgevoerd. 
Bij de eerste stappen, lees ik dit:

 _--- Pulling remote repository for new commits...
./deploy.sh: line 18: git: command not found

 --- Merging/updating checkout.
./deploy.sh: line 24: git: command not found

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated._

vervolgens aan het einde:
 _--- Checking & synchronizing database changes/migrations.
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_notification, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 321 unmodified.

 --- Reloading app code...

 * Reloading Gunicorn (webinterface)...
 [x] Done!

 * Reloading backend process...
 [x] Done!

 * Reloading datalogger process...
 [x] Done!

 --- Clearing cache...
Clearing cache: default
Done

 --- Deployed version:
1.6.2

 >>> Deployment complete. <<<_

Graag je reactie, sinds het dus niet updated naar de 1.7.0 versie.

dennissiemensma, 2017-05-26:
Het lijkt erop dat `git` niet geinstalleerd is. Wat opzich vreemd is, want dat heb je nodig om de application uberhaupt te 'downloaden' bij installatie.

Wil je als sudo/root-user dit proberen: `sudo apt-get install -y git`
En daarna nogmaals proberen te updaten. Git downloadt tevens de laatste versie van de applicatie, dus dat verklaart wel waarom er niets gebeurt.

HFeenstra, 2017-06-04:
He, even laten weten dat deze install git was inderdaad de trick om het te kunnen installeren. Dank!

closed 2017-06-04 by HFeenstra

dennissiemensma, 2017-06-04:
Fijn om te horen :]