dsmrreader/dsmr-reader #326: 3 fase trends en archief data

Bl4ckviper, 2017-05-30:
Dennis, 

Inmiddels heb ik het een goede maand draaien. En ik gebruik vaak de 3 fase grafiek bijna dagelijks om gebruik op een bepaald moment te bepalen. Echter zou ik het graag over de hele maand en dag willen bekijken en vergelijken. 

Dan kan ik kijken of er bepaalde tijdstippen zijn waarin er bepaalde zaken veel gebruiken.

Ook wil ik mijn PV uitbreiden met panelen op mijn schuur die ik dan op een andere fase (die nu het meest verbruikt) aan te sluiten en daarvoor zou die data ook handig zijn. Is dit iets wat binnenkort staat te gebeuren? of kan ik dit al doen op 1 of andere manier?

Alvast bedankt

dennissiemensma, 2017-05-30:
Bedankt voor je verzoek. Momenteel staat dit niet concreet op de planning, echter je kunt deze gegevens wel zelf opvragen via de database. 
Vergelijken is wel wat minder eenvoudig, maar je kunt wel per dag of maand de momenten van het hoogste verbruik (per fase) opvragen. Zie [bijvoorbeeld dit ticket](https://github.com/dennissiemensma/dsmr-reader/issues/161#issuecomment-268869066).

Met deze query kom je denk ik al een heel eind:
```
sudo -u postgres psql dsmrreader -c "select read_at, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3 from dsmr_consumption_electricityconsumption where read_at::text BETWEEN '2017-01-01 00:00:00' AND '2017-02-01 00:00:00' and phase_currently_delivered_l1 is not null order by phase_currently_delivered_l1 desc limit 20;"
```

* De query toont altijd sortering op meeste verbruik van één fase.

* Het stuk `order by phase_currently_delivered_l1 desc` bepaalt wélke fase gebruikt wordt om te sorteren (op hoogste verbruik aflopend). Je kunt hier de `1` veranderen naar `2` of `3` om een andere fase te gebruiken als sortering.

* Het stuk `BETWEEN '2017-01-01 00:00:00' AND '2017-02-01 00:00:00'` bepaalt je selectie. Dit kun je dus op elke willekeurige periode doen, of bijvoorbeeld op een week, dag of zelfs een uur.

* De `LIMIT 20` bepaalt hoeveel resultaten je te zien krijgt.

Leesbaar formaat van de query is overigens:
```
SELECT read_at, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3 
FROM dsmr_consumption_electricityconsumption 
WHERE read_at::text BETWEEN '2017-01-01 00:00:00' AND '2017-02-01 00:00:00' 
    AND phase_currently_delivered_l1 IS NOT NULL
ORDER BY phase_currently_delivered_l1 DESC 
LIMIT 20;
```

Bl4ckviper, 2017-05-30:
dank voor de query.

kan natuurlijk wat data exporteren en zelf wat doen in excel.
Kan ik op de DB ook remote connecten met bijvoorbeeld pgAdmin?

dennissiemensma, 2017-05-30:
Jazeker, alleen mag je daarvoor de (standaard) PostgreSQL configuratie aanpassen. Deze staat meestal geen verbindingen van buitenaf toe.

In `/etc/postgresql/9.4/main/postgresql.conf` (of welk versienummer je ook gebruikt) mag je zoeken naar deze regel:
```
#listen_addresses = 'localhost'
```

Deze staat bij mij rond regel 60 in die file. Je mag hem aanpassen naar iets als:
```
listen_addresses = '*'
```
(Of je eigen IP-adres als deze server op Internet bereikbaar is)

Daarna postgresql herstarten met: ` systemctl restart postgresql` of `service postgresql restart`.

Als het goed is kun je daarna verbinden met de standaard DB user/pass: `dsmrreader` / `dsmrreader`

Bl4ckviper, 2017-05-31:
Hi, 

De qeury werkt top. Heb al kunnen zien welke fase nu het meeste verbruikt heeft deze maand.

Het remote connecten is helaas nog niet gelukt krijg deze melding:

Unable to connect to server:

FATAL: no pg_hba.conf entry for host "<iplaptop>", user "dsmrreader", database "dsmrreader", SSL on
FATAL: no pg_hba.conf entry for host "<iplaptop>", user "dsmrreader", database "dsmrreader", SSL off

Bl4ckviper, 2017-05-31:
Heb het al: 
host  all  all 0.0.0.0/0 md5 toegevoegd aan de /etc/postgresql/9.4/main/pg_hba.conf

dennissiemensma, 2017-05-31:
Top, goed om te horen! Die HBA was ik even vergeten, fijn dat je het zelf zo snel hebt gevonden.

Bl4ckviper, 2017-05-31:
Even een controle vraag neem aan dat het waarde zijn in watt dus 6.771 watt is 6,771 kWh?

dennissiemensma, 2017-05-31:
Ja en nee. De getallen per fase zijn in kW (vermogen) en niet in kWh (waarvoor je betaalt). Dus 6771 Watt of 6,771 Kilo Watt.

dennissiemensma, 2017-09-29:
Ik hoop dat je inmiddels via de queries alle benodigde info hebt kunnen vinden voor je eigen doelen. En anders kan ik je daar nog wel verder mee proberen te helpen.

Uiteindelijk heb ik besloten hier (voorlopig) geen ondersteuning voor te maken in de interface. Voor nu sluit ik deze.

closed 2017-09-29 by dennissiemensma