dsmrreader/dsmr-reader #328: Dag/uurtotalen worden niet correct hergegenereerd na Delete Aggregated Data script

Ernst79, 2017-06-06:
Allereerst bedankt voor het vele werk wat je hierin gestopt hebt!

Helaas ben ik denk ik tegen een bug aangelopen en heb aan het einde nog een paar verbeterpuntjes, ter overweging. Eerst even een toelichting. Ik ben bezig mijn oude metingen over te zetten van InfluxDB naar DSMR-reader. Hiertoe heb ik een aantal meetpunten (1-5 t/m 4-5) via de API v2 geupload naar DSMR-reader, wat prima gaat (als test eerst 4 dagen gedaan). Daarnaast zijn er ca 4 dagen aan metingen van DSRM-reader zelf aanwezig (1-6 t/m heden). Zoals in de notitie omschreven staat heb ik daarna ./manage.py dsmr_backend_delete_aggregated_data gedraaid, waarna DSMR-reader alle gegevens opnieuw gaat aggregeren. 

> Daarom kan het nodig zijn dat je alle geaggregeerde gegevens verwijdert met het ./manage.py dsmr_backend_delete_aggregated_data commando.

Nadat alles weer ingelezen en geaggregeert is, zie ik dat de dagtotalen van de via de API geuploade dagtotalen (1/5 - 4/5) correct in DSMR-reader zijn gekomen. Echter, de data die door DSMR-reader de afgelpen dagen zelf zijn uitgelezen uit de slimme meter (1/6 - 4-6), zijn niet meer correct. 

[image]

Zoals je kan zien zijn de waarden van zowel gas als electra veel te laag. De teruglevering is zelfs helemaal 0 (terwijl de zonnepanelen overuren draaide afgelopen weekend). Bij gas en electra gebruik zijn de waarden zeer laag. Overigens waren de waarden voordat ik het  ./manage.py dsmr_backend_delete_aggregated_data commando draaide wel correct. Ook de data van vandaag is weer correct. 

Heb je enig idee wat hier mis gaat?

Alvast dank voor je hulp. 

NB. Misschien kan je aan de handleiding toevoegen dat je minimaal twee metingen op een dag moet uploaden om een dagtotaal te genereren. 
NB.2. de API verplicht het om 'electricity_currently_delivered' en 'electricity_currently_returned' mee te sturen. Ik ben voor het verleden alleen geintereseerd in de dagopbrengsten, dus zie de meerwaarde van de kW waarden niet zo. Leeglaten mag echter niet. Heb het nu maar opgelost door een gemiddelde van die dag (kwh / 24 h) mee te sturen. 0 kW kan natuurlijk ook. Wellicht overwegen om deze niet verplicht te stellen.

dennissiemensma, 2017-06-06:
Bedankt voor je uitgebreide melding!

Ik heb niet zo 1-2-3 een idee waar dit aan kan liggen. Het is wel erg toevallig dat het precies de dagen betreft die je initieel had gemeten. Vermoedelijk zit er dan nog een bugje in het verwerken met terugwerkende kracht.

* Voor de duidelijkheid: Alle gegevens die je gemeten en ingevoerd hebt, betreffen dezelfde slimme meter?

* Wat betreft `electricity_currently_delivered` en `electricity_currently_returned`. Je mag die in dat geval op `0` zetten, zoals je zelf al aangeeft. De API kán gebruikt worden voor historische metingen, maar is er niet primair voor gemaakt (Dashboard gebruikt o.a. deze gegevens).

* Ik zal later kijken of ik de docs kan bijwerken met bovenstaande punten. Goede aanvulling

Ernst79, 2017-06-06:
Ja, het betreft dezelfde meter. De meterstanden zijn oplopend (als je daarop doelt). Het betreft een DSMR v2 meter overigens. Kan het daar iets mee te maken hebben?

Ik zal morgenavond ook wel eens in het python script duiken. Kijken of ik iets kan ontdekken. Het gekke is dat de via de API ingevoerde gegevens wel correct zijn, terwijl die door het programma zelf zijn uitgelezen niet (terwijl ze dat voor het script wel waren).

dennissiemensma, 2017-06-07:
Uiteindelijk zou dat niet mogen uitmaken. 

Je zou eens kunnen kijken in de tabellen `dsmr_consumption_gasconsumption` (splitsing van meting naar scheiding electra/gas. En in `dsmr_stats_hourstatistics` (aggegratie van de eerdergenoemde tabel, gebundeld op uren. Wellicht zie je daar nog iets opvallend.

Ernst79, 2017-06-07:
Ik heb net al mijn oude metingen overgezet en ben de boel weer opnieuw aan het hergenereren. Ik heb deze keer vooraf in systemctl het uitlezen van de meter even tijdelijk uitgezet, totdat alles hergegenereerd is. Wellicht dat dat verschil maakt, al betwijfel ik het. 

Helaas moet ik dus een paar uurtjes wachten tot alles weer hergegenereerd is. Daarna zal ik eens in de tabellen kijken of ik wat bijzonders te zien is.

Ernst79, 2017-06-08:
Het werkt!. Wat ik heb gedaan is het volgende:

1. Oude meterstanden geupload via de API v2, 2 meterstanden per dag (begin en einde v/d dag)
2. de dsmr_datalogger tijdelijk stopgezet
`sudo supervisorctl`
`stop dsmr_datalogger`
3. het script gedraaid om alle geaggregeerde gegevens te verwijderen
`./manage.py dsmr_backend_delete_aggregated_data` 
4. de gegevens worden vervolgens weer gehergenereerd. Nadat alles gereed was (duurde enkele uren, zie statuspagina) heb ik de datalogger weer aangezet.
`sudo supervisorctl`
`stop dsmr_datalogger` 

Ik vermoed dat het door laten lopen van de data_logger tijdens het opnieuw aggregeren van de data dus roet in het eten heeft gegooid.

dennissiemensma, 2017-06-08:
Top, goed om te horen dat dat het uiteindelijk oplost!

Ik laat het ticket nog even open om dit later in de documentatie bij te werken. Bedankt :]

dennissiemensma, 2017-06-08:
**Todo: Docs bijwerken**

dennissiemensma, 2017-06-13:
Ik heb het bijgewerkt, nogmaals bedankt.

closed 2017-06-13 by dennissiemensma