dsmrreader/dsmr-reader #332: Add info to FAQ

balk77, 2017-06-16:
Hi, everytime I want to update DSMR reader (happy user by the way!) I need to log in as dsmr user, and every time I need to look into the installation docs how to do the `sudo su - dsmr`.

Could you, under [How can I update my application](https://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application), add  `sudo su - dsmr` near `Make sure to execute it while logged in as the dsmr user:`? Thanks!

WatskeBart, 2017-06-16:
You could create a new alias to make it easier.
Edit your **.bashrc** file of the current user (not the dsmr user) and add the following under the other alias lines: `alias dsmr='sudo su - dsmr'` (afterwards run `source .bashrc` to reload your .bashrc file)

This way you only need to type `dsmr` and you will go straight into the dsmr user 👍 

You can use the command **vi** to edit the .bashrc file or **nano** which I find more user friendly
(you may need to install the nano package `sudo apt-get install nano`).

dennissiemensma, 2017-06-16:
Thank you for reporting, I will adjust the docs a bit for the next release.

dennissiemensma, 2017-09-11:
I have added the `sudo su - dsmr` as requested, for the next release.

closed 2017-09-11 by dennissiemensma