dsmrreader/dsmr-reader #333: Daily Cost calculations

MJBBRU, 2017-06-16:
Something is wrong with the calculation of the daily costs for the electricity.
I entered a price of 0.16661 per kWh.
I used 2.66 kWh but the costs show as €0.15.
Refreshing the page or restarting all the services don't make any difference.

Thanks in advance,

dennissiemensma, 2017-06-16:
Thank you for reporting this issue. I do need some more information, since I've not experienced this myself yet.

* Can you make a screenshot of it?

MJBBRU, 2017-06-17:
Here is a screenshot after killing all processes and rebooting the Pi.
It's still not working properly.

[image]

dennissiemensma, 2017-06-17:
Thank you for providing the information. I'm not sure how this could happen, some possibilities:

- Did you make sure the start and ending date of the energy prices are within this month?

- You could have the application recalculate all readings again, by deleting the aggregated data. This will **not** affect the readings (source data):
```
sudo su - dsmr
./manage.py dsmr_backend_delete_aggregated_data
```

If you just started using the application, this process will delete the data quickly and I presume the background process will regenerate the data quite fast as well. 
More info here, in the 'Note' block: https://dsmr-reader.readthedocs.io/en/latest/api.html#id2

Let me know if this fixes your problem. If the prices are set correctly, a clean sweep should show valid data.

MJBBRU, 2017-06-23:
Today I was finally able to check this.
- The start and ending date for the energy prices are correct. (01-2017 to 01-2018)
- Recalculation of the readings with the command line you sent didn't work

An error showed up that the commandline needs to be extended with  --ack-to-delete-my-data 
(or something)
By doing this I think that all data is gone :(
I killed all processes and rebooted the raspberry but I still get this screen.
[image]

For now I will see how the calculations are going.
I'll keep you informed.

Thanks so far

dennissiemensma, 2017-06-23:
I see I forget to mention the 'ack' parameter. It's to prevent people from randomly trying commands.

You can safely execute the `./manage.py dsmr_backend_delete_aggregated_data --ack-to-delete-my-data` command. 
It will **not** delete the readings (source data), but it will delete all data calculated from the readings, such as the day totals and costs. It will basically reset all stats and start from the beginning, while using the readings stored so far.

See also #328 and [this comment](https://github.com/dennissiemensma/dsmr-reader/issues/328#issuecomment-307013093) for more info.

dennissiemensma, 2017-07-02:
@MJBBRU did my last comment fix your issue?

MJBBRU, 2017-07-05:
Everything is working fine now.
Thanks!

[image]

closed 2017-07-05 by MJBBRU