dsmrreader/dsmr-reader #334: opnieuw opbouwen DSMR

HFeenstra, 2017-06-17:
zcat <temp/dsmrreader-POSTGRESQL-BACKUP-Friday.sql.gz> | sudo sudo -u postgres psql dsmrreader
-bash: syntax error near unexpected token `|'

Bovenstaande probeer ik uit te voeren nadat mijn net nieuwe SD kaart is gecrasht, kan dus opnieuw beginnen. Wil vervolgens een backup inlezen, zie bovenstaande eerste regel, heb het bestand dus via SMB vanaf een Windows PC naar een folder gebracht die "temp" heet.

Krijg dan dus volgende regel als retour. Kortom ik krijg de backup niet ingelezen. Wat doe ik fout, of wat is er fout aan de regel om backup in te lezen. Ik volg de stappen voor installatie en ben dus aan het einde van stap 1 waar het gaat over het inlezen van een backup bij een nieuwe setup!

helmo, 2017-06-17:
De < en > moet je weglaten uit het commando. Die zijn enkel in de documentatie bedoeld om aan te geven dat het een verplichte parameter is.

dennissiemensma, 2017-06-17:
@HFeenstra klopt het dat je het commando uitvoert inclusief `<` en `>`? Zo ja, dan hoort dat inderdaad niet. Dat is scherp gezien van helmo.

Een voorbeeld van hoe ik het uitvoer, wanneer ik in dezelfde map zit: 
`psql -f dsmrreader-postgresql-backup-Saturday.sql dsmrreader`
(dus zonder die tekens, alleen gaat mijn voorbeeld over een ongecomprimeerd bestand)

HFeenstra, 2017-06-18:
Top Beiden! Dank voor de terugkoppeling. Ik heb het uiteindelijk weer aan de praat gekregen. Heb vanuit de folder het bestand zoals jij @dennissiemensma beschreef uitgevoerd.
De rest lukte weer prima. Mooi dat door de backup terug lezen ook alle andere instellingen weer terug zijn.

dennissiemensma, 2017-06-18:
Top, goed om te horen.

closed 2017-06-18 by dennissiemensma