dsmrreader/dsmr-reader #335: API Installatie handleiding.

Mixpower, 2017-06-25:
Hallo,

Ik ben echt super blij met DSMR Reader en zeer onder de indruk van alle functionaliteiten die hierin verwerkt zijn!
Helaas is mijn Pi Zero W niet krachtig genoeg om DSMR Reader standalone te draaien,
daarom wou ik gebruik gaan maken van de API om zo de telegrams vanaf de P1 bij de Slimme meter door te sturen naar DSMR Reader draaiende in een VM op ESXI.

Maar nu kom ik er niet uit hoe ik dit moet gaan opzetten en installeren op de PI,
er word in de configuratie van de API bijv verwezen naar "/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrclient/bin/python3" maar ik heb geen idee hoe ik dit moet opzetten en wat hier precies voor nodig is.

Het zou fijn zijn als hier een handleiding voor zou komen, of wat meer duidelijk over hoe de API functie op te zetten en in te stellen voor remote met DSMR Reader.

Alvast bedankt, en ga vooral zo door! :)

trbs, 2017-06-25:
You should be able to run only the `dsmr_datalogger` task on the PI Zero then you can run the rest on a VM.

This would be the most lightweight way and the PI only needs to be able to access PostgreSQL besides having the P1 connected. (For this you might need to change your `pg_hba.conf` depending on how that's setup)

For this to work you should be able to follow the instructions for the full setup on the VM but exclude `program:dsmr_datalogger` from the `supervisord` config. On the PI skip all the setup for the website (things like nginx, collectstatic, etc) and comment out `program:dsmr_webinterface` and `program:dsmr_backend` in the supervisord setup.

The database configuration in `settings.py` might change too on the PI since you now point it to a PostgreSQL server running on your VM.

If your familiar with Python and VirtualEnv's ( https://docs.python.org/3/library/venv.html#module-venv ) you should be able to setup `dsmr-reader` anywhere on your system and adjust the values from the documentation accordingly.. if your not familiar with Python or Linux it's probably best to stick closely to the documentation :)

Mixpower, 2017-06-26:
Thank you for your reply!
I can install everything on the PI except the Nginx webserver, but I still don't know how to point the Pi to the PostgresSQL on the VM, also how do I exclude stuff from supervisord exactly?

I do understand what you are trying to say, but it is still very unclear how to setup this all correctly,
that's why a manual of some sort would be a great addition to this API.

I hope the developer of DSMR Reader can shed some light on this.

dennissiemensma, 2017-06-26:
Bedankt voor je ticket. Ik probeer de API-documentatie bewust wat 'simpel' te houden, eigenlijk vooral voor ontwikkelaars, maar ik zal je proberen zoveel mogelijk te helpen. 

* Als eerste zul je DSMR-reader ergens anders moeten installeren, bijvoorbeeld op een VM, zoals je al aangeeft. Is dat al gelukt?

* Verder is het in principe niet meer dan wat [hier staat beschreven](http://dsmr-reader.readthedocs.io/nl/latest/api.html#api-v1-remote-datalogger):

```
sudo apt-get install -y supervisor python3 python3-pip python3-virtualenv virtualenvwrapper
sudo useradd dsmr --home-dir /home/dsmr --create-home --shell /bin/bash
sudo usermod -a -G dialout dsmr
sudo su - dsmr
mkdir ~/.virtualenvs
virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3

# Zet deze twee commands onderin /home/dsmr/.bashrc
source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader

# Vervolgens weer terug naar commandline als dsmr-user:
source ~/.virtualenvs/dsmrreader/bin/activate
pip install pyserial==3.2.1
pip install requests==2.12.4

# Kopieer het script uit de docs-link naar:
# '/home/dsmr/dsmr_datalogger_api_client.py'.

# Vergeet niet om de volgende variabelen in het script te vullen met je eigen hostnaam van je VM + de API key uit de admin-interface van dsmrreader:
API_SERVERS = (
    ('http://HOST-OR-IP-ONE/api/v1/datalogger/dsmrreading', 'APIKEY-BLABLABLA-ABCDEFGHI'),
)

# Nu weer terug naar root/sudo-user:
# Maak '/etc/supervisor/conf.d/dsmr-client.conf' en zet dit erin
[program:dsmr_client_datalogger]
command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrclient/bin/python3 -u /home/dsmr/dsmr_datalogger_api_client.py
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=20
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=10MB
stdout_logfile_backups=3

# Nu starten met:
sudo supervisorctl reread
sudo supervisorctl update

# Als het goed is zie je nu 'dsmr_client_datalogger' op RUNNING staan met:
sudo supervisorctl status

```

Dit is allemaal gekopieerd/uit mn hoofd, wellicht mis er nog wat. Ik hoor graag van je of het lukt.

Mixpower, 2017-06-26:
Super bedankt daarvoor, ik ga er meteen mee aan de slag!
Moet op de VM de datalogger nog uitgeschakeld worden of is dat niet nodig?

dennissiemensma, 2017-06-26:
Ik zou die wel uitschakelen gezien het voor onnodige IO zorgt. 

Je kunt daar de Supervisor-config bewerken en het hele blok onder (en inclusief) `[program:dsmr_datalogger]` uit te commenten met `#`'s.

Vervolgens reloaden als root/sudoer met:
```
sudo supervisorctl reread
sudo supervisorctl update
```

Als het goed is verdwijnt het proces uit `sudo supervisorctl status`.

Mixpower, 2017-06-26:
Top, dat is gelukt en het draait helemaal zonder problemen op de VM!
Nu ga ben ik bezig met het installeren op de Pi Zero W, maar als ik het goed begrijp moet ik nog wel dsmr reader via de github binnen halen? 

Aangezien dit moet toevoegen in bashrc
``source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader``

Mixpower, 2017-06-26:
Ik heb het nu precies zo geinstalleerd op de PI zoals je hebt aangegeven maar ik krijg deze melding.

dsmr_client_datalogger           BACKOFF    Exited too quickly (process log may have details)

En als ik dan "tail -f dsmr_client_datalogger" doe krijg ik helemaal niks.
Dus ergens gaat er iets niet goed.

Mixpower, 2017-06-26:
Ik heb de fout al gevonden!

In het script wat je gebruikt verwijs je naar "/home/dsmr/.virtualenvs/dsmrclient/bin/python3" maar dsmrclient bestaat niet in de virtualenv aangezien je aangeeft "virtualenv ~/.virtualenvs/dsmrreader" te maken.

Dus ik heb nu dsmrclient aangepast naar dsmrreader en nu werkt het allemaal perfect!
De VM krijg readings binnen vanaf de meter, en dit had ik eigenlijk eerder ook al zover maar kwam ik op hetzelfde uit als nu.

Dus daar zit de hele fout in waarom de API niet werkte! :)
Super bedankt voor de hulp!

dennissiemensma, 2017-06-27:
Bedankt voor al je terugkoppeling en mooi dat je er zelf uit bent gekomen!

 Ik zal die typo van 'dsmrclient' aanpassen, want dat klopt inderdaad niet. Zelf heb ik daar overheen gelezen. :]

closed 2017-06-27 by dennissiemensma

rodeho, 2019-04-27:
@dennissiemensma hieronder is de meest basale setup geschetst, heb ik hier in iets gemist? De Pi hoeft alleen maar de client datalogger te draaien en de VM de overige processen.

Nog tips voor updates in de toekomst, of heeft de Pi nagenoeg geen updates nodig omdat hij slechts de data doorgeeft via het python script?

VM:
user@VM:/home/dsmr $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 14101, uptime 3:30:46
dsmr_mqtt                        RUNNING   pid 5563, uptime 0:00:15
dsmr_webinterface                RUNNING   pid 14099, uptime 3:30:46

pi@raspberrypi:/home/dsmr $ sudo supervisorctl status
dsmr_client_datalogger           RUNNING   pid 8312, uptime 2:26:56

dennissiemensma, 2019-04-28:
Ja in principe is dat een van de twee manieren om dit te doen. En je hoeft je Pi dan geen updates te gevenm gezien die alleen `PySerial` en `requests` library gebruikt. Dus poort uitlezen en doorsturen via HTTP, meer doet die eigenlijk niet.

rodeho, 2019-04-29:
Dennis misschien even je post wijzigen er staat nog steeds client in, in de formele guide is het wel goed. 

`command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/**dsmrclient**/bin/python3 -u /home/dsmr/dsmr_datalogger_api_client.py`