dsmrreader/dsmr-reader #340: Show combined low+high tariff value

helmo, 2017-07-01:
It would be nice to have the sum of the low and high tariff on both the dashboard and the archive pages.
Not that the math is that difficult ;) but it's just easier to compare the daily or monthly usage that way.

Here's a mockup:
[image]

From a usability perspective it might be a bit confusing combined with the 'Total' row ... I'm open to suggestions.

dennissiemensma, 2017-07-02:
Thank you for your suggestion. Did you also try [merging the tariffs](https://dsmr-reader.readthedocs.io/en/latest/settings.html#merge-electricity-tariffs)?

 It's not quite the same, but it sums up the totals everywhere. It's a interface-only feature, so it shouldn't affect any data read and can be turned off again any time.

helmo, 2017-07-02:
An html change was enough ... while experimenting a bit I ended up with:[image]

helmo, 2017-07-02:
The merge function is a reasonable workaround ... but I'm interested in both.

dennissiemensma, 2017-07-02:
Merged #341

closed 2017-07-02 by dennissiemensma