dsmrreader/dsmr-reader #342: Backup to dropbox never finish (free plan no more space)

joopy, 2017-07-06:
Hi,
I have version 1.8 running. My dropbox account was full, no more free space. The DSMR program keeps trying to backup (+- 30MB). So in 1 day I got a alarm from my monitoring software that my raspberry was using 50GB bandwith a day. Networkscanner reveals my raspberry was doing dropbox request.

Can you build in a check that when dropbox account is full, backup will be paused?

goegol, 2017-07-06:
Already upgrade to version 1.8.1?
There was an issue with the dropbox client in dmsr.

dennissiemensma, 2017-07-06:
Thank you for reporting this! The application does not check the quota available, but it definitely should. I will fix this in a future release.

dennissiemensma, 2017-07-06:
The v1.8.1 patch was only for #339, due to a heads up of Dropbox that the API will change in two months, so unrelated to this issue.

joopy, 2017-07-08:
Thnx Dennis

dennissiemensma, 2018-03-13:
Even lokaal getest, ik kan een melding maken op basis van een ruwe matching op `insufficient_space` in de fout:
```
ApiError('x', UploadError('path', UploadWriteFailed(reason=WriteError('insufficient_space', None), upload_session_id='x')))
```

En dan tevens de backups pauzeren voor een (halve) dag.

joopy, 2018-03-14:
Lijkt mij prima Dennis!

dennissiemensma, 2018-03-15:
Gemerged naar `development` voor de volgende release.

closed 2018-03-15 by dennissiemensma

dennissiemensma, 2018-03-21:
Uitgebracht in [v1.15](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.15.0).