dsmrreader/dsmr-reader #343: Teruggeleverd wordt niet meegenomen in kosten

vgouw, 2017-07-07:
De Kosten (€) zoals weergegeven bij Elektriciteit 2 (piektarief) houd geen rekening met het Teruggeleverd (kWh) . Ik zou verwachten dat hier een negatief bedrag zou staan ipv van 0,10€
[image]

dennissiemensma, 2017-07-08:
Bedankt voor het melden van dit issue. Het is overigens een bekend iets, er staat ook al #163 voor om het terugleveren uberhaupt goed te verwerken. Wegens de salderingsregeling en verschillende teruglevertarieven, kan ik er (nog) niet van uitgaan dat dezelfde prijs gebruikt wordt voor teruglevering. Daarvoor is meer nopig.

Ik raad je aan om #163 in de gaten te houden. Deze staat overigens niet voor de korte termijn op de planning.

closed 2017-07-08 by dennissiemensma

dennissiemensma, 2017-09-28:
Opgepakt in #163. Zie hier hieronder voor een voorbeeld met fictieve getallen en (onrealistische) bedragen:

[image]