dsmrreader/dsmr-reader #344: dsmr-reader in HASS custom panel

nelbs, 2017-07-14:
Allereerst mijn complimenten voor de toffe applicatie die je hebt gemaakt!

Ik zou graag de dsmr reader onder willen brengen in een custom panel of Iframe in Homeassistant. Op de een of andere manier werkt een simpel Iframe niet. Voor een custom panel lijken de mogelijkheden wat ruimer.  Hiervoor dient een pad naar de webcomponent aangegeven te worden zoals dit voorbeeld:

    webcomponent_path: /home/hass/hello.html

Is er een ingang van dsmr die ik hiervoor kan gebruikten of denk je dat dit niet mogelijk zal gaan?

alvast bedankt!

dennissiemensma, 2017-07-15:
Bedankt voor je vraag. Dit lijkt te komen door de standaard [`X-Frame-Options`](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/X-Frame-Options) header, die het onderliggende framework van dit project (Django) meegeeft. 
Waarschijnlijk geeft je browser, als je op F12 drukt, de volgende foutmelding: 
`Refused to display 'http://localhost:8000/' in a frame because it set 'X-Frame-Options' to 'sameorigin'.`

Je kunt dit uitzetten door het volgende toevoegen aan DSMR-reader in `dsmrreader/settings.py`:

`X_FRAME_OPTIONS = 'ALLOWALL'`

Het is echter beter om alleen de URL op te geven die toegang mag hebben tot je applicatie. Dus de URL waarop je via je browser in HASS het iFrame gaat bekijken. Die kun je dan opgeven als waarde:

`X_FRAME_OPTIONS = 'ALLOW-FROM https://example.com/'`

Aanpassen:
```
sudo su - dsmr
vi dsmrreader/settings.py

# Voeg onderstaande regel toe.
X_FRAME_OPTIONS = 'ALLOWALL'

# Sla bestand op.

# Herlaad de applicatie met:
./reload.sh
```

Ik hoor graag of je hiermee de iFrame werkend krijgt.

nelbs, 2017-07-15:
Dit werkt! Thanks

closed 2017-07-15 by dennissiemensma