dsmrreader/dsmr-reader #346: Vastlopen dashboard door hoge aantal database records?

pjeterinfo, 2017-07-26:
In een postgres database op Raspberry3 (iscsi disk) lees ik een L&G DSM5 uit.
Er is ingesteld om de telegrammen elke seconde te verwerken, groeperen werkte niet dan worden de telegrammen helemaal niet verwerkt.
Nu staan er na ruim 2 maanden zo'n 1,5 tot 2 miljoen regels in de database, de statistieken pagina laad het aantal metingen en min en max verbruik niet meer.
Ik vraag mij af of het aantal records in de database het vastlopen en niet meer kunnen laden van de statistieken kan veroorzaken?

dennissiemensma, 2017-07-26:
Bedankt voor je ticket. De statistieken laden wel, maar doen dat op de achtergrond. Je zult daarvoor even geduld moeten hebben wanneer de pagina geopend is. Om de een of andere reden kan PostgreSQL geen snelle count() doen op de database.
Ik zit zelf op 5 miljoen metingen en het duurt dan zo'n 20 seconden voordat ze zichtbaar zijn. 

Wat betreft dat de telegrammen niet gegroepeerd worden, hoe bedoel je dat? Is er dan een achterstand wanneer je de Status-pagina bekijkt? Ik heb zelf helaas een DSMR 4 meter, dus ik kan het hier niet reproduceren voor een gelijke situatie qua hoeveelheid telegrammen.

pjeterinfo, 2017-07-27:
Waarschijnlijk kom ik dan wat iops te kort om dsmr-reader te kunnen draaien, ik krijg namelijk een gateway timeout uiteindelijk op de statistieken pagina. 
Bij het installeren wilde ik de metingen instellen om per 10 seconden te verwerken omdat ik al het idee had dat elke seconde heel veel records oplevert, in dat geval loopt het aantal onverwerkte telegrammen alleen maar op, er wordt niets verwerkt. Niet heel lang getest , maar toch zeker enkele minuten.
Ik zie wel wat in het kunnen instellen van retentie.

dennissiemensma, 2017-07-27:
Je zou nog kunnen kijken met `top` of het systeem constant op I/O wacht. 

Voor rententie is wel een ticket (#72), alleen staat die momenteel nog niet gepland.

Varsame, 2017-07-28:
Programmeren is slechts een zijdelingse hobby dus misschien zeg ik wel wat raars, maar waarom zou je voor de statistieken elke keer bij het laden ervan de hoeveelheid entries willen _tellen_? Je kunt toch de laatste of hoogste counter/entry ID oid opzoeken en weergeven? Scheelt een hoop resources en laadtijd lijkt mij..

dennissiemensma, 2017-07-29:
Dat klopt. Op het moment van implementeren was het nog prima te doen qua snelheid, echter heeft PostgreSQL een zwakte met de (exacte) count op grote tabellen. 
Het ophalen van het laatste ID is een mooie workaround, maar op het moment dat retentie ingebouwd wordt, mag dat wéér over de kop (het klopt anders niet meer). De statistiekenpagina is overigens nooit bedoeld om telkens te bekijken, het is meer een nice te have. Wellicht dat ik op termijn dit nog aanpak, of de count op het aantal metingen lostrek van de andere statistieken.

Jeltel, 2017-08-05:
Ik moet zeggen dat ik ook tegen dit probleem aanloop. Er kan maar één pagina geladen worden. Een volgende pagina levert een gateway timeout op.

Ik dacht dat het aan de pi lag met allerlei zaken die daar ook op stonden (pihole en nextcloud). Na het opnieuw installeren en het terugzetten van een backup is het helaas niet sneller geworden.

Ik zit inmiddels op iets meer dan 2 miljoen metingen.

trbs, 2017-08-05:
Zou interessant zijn om te weten of index-only count(*) sneller zouden zijn.
Vaak kiest PostgreSQL standaard voor een seq-table scan.

```
dsmrreader=# explain SELECT COUNT(*) AS "__count" FROM "dsmr_datalogger_dsmrreading";
                                        QUERY PLAN
------------------------------------------------------------------------------------------
 Aggregate  (cost=13052.71..13052.72 rows=1 width=8)
   ->  Seq Scan on dsmr_datalogger_dsmrreading  (cost=0.00..11825.57 rows=490857 width=0)
(2 rows)

Time: 0.275 ms
```

Zou iemand die hier last van heeft het volgende op hun database kunnen draaien ?

```
dsmrreader=# \timing on
Timing is on.
dsmrreader=# SELECT COUNT(*) AS "__count" FROM "dsmr_datalogger_dsmrreading";
 __count
---------
  499856
(1 row)

Time: 57.459 ms
dsmrreader=# set enable_seqscan = off;
SET
Time: 0.162 ms
dsmrreader=# SELECT COUNT(*) AS "__count" FROM "dsmr_datalogger_dsmrreading";
 __count
---------
  499857
(1 row)

Time: 93.224 ms

dsmrreader=# explain analyze verbose SELECT COUNT(*) AS "__count" FROM "dsmr_datalogger_dsmrreading";
                                                                                             QUERY PLAN
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 Aggregate  (cost=17995.42..17995.43 rows=1 width=8) (actual time=909.127..909.128 rows=1 loops=1)
   Output: count(*)
   ->  Index Only Scan using dsmr_datalogger_dsmrreading_9d090a40 on public.dsmr_datalogger_dsmrreading  (cost=0.42..16768.28 rows=490857 width=0) (actual time=0.034..501.079 rows=499866 loops=1)
         Output: processed
         Heap Fetches: 106692
 Planning time: 0.086 ms
 Execution time: 909.175 ms
(7 rows)
```

In mijn geval met een half miljoen rijen is de sequential scan sneller maar het maakt niet zo gek veel uit. Op systemen met veel meer rijen en/of IO en/of minder geheugen is er goeie kans dat de index scan wel sneller is.

Als dat zo is dan kan het een idee zijn om PostgreSQL te dwingen om de index scan te doen voor deze query's.

dennissiemensma, 2017-08-05:
@Jeltel gaat het bij jou over de hele applicatie? Bij mij (op Pi 3) is alleen de statistiekenpagina traag, en dan puur de kopjes `TELEGRAM-METINGEN` / `VERBRUIKSSTATISTIEKEN`, die op de achtergrond laden. Het kan zijn dat je browser wacht op het laden daarvan, wanneer je daarna doorklikt naar een andere pagina.

Jeltel, 2017-08-05:
Hele applicatie. Als eerste open ik het dashboard, dat gaat traag, maar
laad uiteindelijk wel.
Klik ik daarna op een andere link, krijg ik eigenlijk altijd een gateway
timeout.

pjeterinfo, 2017-08-06:
Ik was iets te optimistisch wb het aantal records ;-) het zijn er ruim 7 miljoen inmiddels.
Het vastlopen is de laatste tijd niet meer voorgekomen, en de gui werkt zolang de statistieken niet geopend worden.
In de configuratie instellingen loopt de gui vast met een gateway timeout als ik de DSMR-uitlezing open.

dsmrreader=# \timing on
Timing is on.
dsmrreader=# SELECT "dsmr_consumption_electricityconsumption"."currently_delivered", COUNT("dsmr_consumption_electricityconsumption"."currently_delivered") AS "currently_delivered_count" FROM "dsmr_consumption_electricityconsumption" WHERE "dsmr_consumption_electricityconsumption"."currently_delivered" > 0 GROUP BY "dsmr_consumption_electricityconsumption"."currently_delivered" ORDER BY "currently_delivered_count" DESC LIMIT 5
;
 currently_delivered | currently_delivered_count 
---------------------+---------------------------
               0.143 |                       569
               0.142 |                       552
               0.144 |                       494
               0.146 |                       404
               0.141 |                       378
(5 rows)

Time: 334.512 ms
dsmrreader=# SELECT COUNT(*) AS "__count" FROM "dsmr_datalogger_dsmrreading";
 __count 
---------
 7199286
(1 row)

Time: 70852.990 ms
dsmrreader=#

dennissiemensma, 2017-08-12:
Ik heb van de week nog even gekeken of ik het kan verbeteren, maar het lijkt erop dat het een beperking is van PostgreSQL's systeem onderwater. Counts op grote tabellen op trage hardware (zoals een Pi), zijn daardoor niet te optimaliseren en traag.

Daarom heb ik net een patch/update uitgebracht, waarin het opvragen van deze trage statistieken een handmatige actie is geworden. Het openklikken van de statistieken-pagina lijdt dan niet automatisch tot de traagheid. Ik had gehoopt een betere oplossing te maken, maar helaas.

closed 2017-08-12 by dennissiemensma