dsmrreader/dsmr-reader #347: DSMR Reader op externe server

Kevintjeb, 2017-08-10:
Beste Dennis,

Wat een geweldig product heb je hier! Ik heb een vraag wat betreft de DSMR Reader op een externe server draaien.

Ik heb alles nu geïnstalleerd op mijn server, draait ergens in Amsterdam, en wil deze alle date voeden door puur de API te gebruiken. Ik heb dan ook een orange pi zero (ik verwacht dat die te zwak is om alles te draaien, daarom doe ik het zo) die data van de slimme meter polled en dat door gaat sturen naar de API. Nu wil ik er dus voor zorgen dat de datalogger niet aan staat.

Ik heb in de settings de optie 'poll P1 poort' uitstaan, er staat echter dat dit niet de bedoeling is en het er ook uitgehaald wordt in een latere release. 

Nu is mijn vraag: wat is de correcte manier om alles extern te draaien en de data te voeden via te API?

dennissiemensma, 2017-08-10:
Ik kan je sowieso aanraden om het op een server te draaien, qua performance, dus dat is mooi. Het pollen van de P1 poort zal nooit verdwijnen, omdat dat gebruikt wordt voor de datalogger op de Pi. 

Het makkelijkste is om [het script voor de v1 API](http://dsmr-reader.readthedocs.io/nl/latest/api.html#script) te gebruiken. Deze kun je dan in Supervisor op je Orange Pi draaien. Vergeet niet de URL naar je server + API key in het script te zetten.

Kevintjeb, 2017-08-11:
Yes, het werkt nu heel goed. Het is inderdaad beter om het op een server te draaien, die zijn gewoonweg krachtiger. 

Mede doordat ik het nu op een veilige manier aan het internet kon blootstellen en het krachtiger was, heb ik het gedaan.

Ik had zelf ook een script gemaakt om de data te uploaden en dat werkt nu allemaal prima. Een opmerking wel, de link naar je readTheDocs beschrijft een script dat ergens in de virtualenv staat. Dat is allemaal prima alleen als je de 2 componenten splitst van elkaar dan heb je dus al die bestanden etc niet op je (bijvoorbeeld orange pi) data poller staan. 

Voor de mens die het wat simpeler wilt hebben is het misschien handig om een apart stukje daarvoor te vermelden (puur, installeer python3, virtualenv? En start het p1-poll script).

dennissiemensma, 2017-08-11:
Oke goed om te horen dat het gelukt is! Ik laat de docs even voor wat ze zijn, het is meer een voorbeeld voor de wat geavanceerde scripters. Toch bedankt voor de tip! :]

closed 2017-08-11 by dennissiemensma