dsmrreader/dsmr-reader #349: Rebuild of consumption and statistics inomplete

kvolleberg, 2017-08-19:
Hi Dennis,
I have been running the DSMR-reader for 8 months now (logging with 10sec interval).
Last week I noticed that the status update caused a hanging app. First I thought that my hanging app was due to the large database, but 2.something million readings is not that big... 

I found out that it had to do with several duplicate IDs in the ElectricityConsumption table, resulting in an error and a hanging application.
Therefore I used manage.py dsmr_backend_delete_aggregated_data to delete all statistics and consumption data (sidenote: check the https://dsmr-reader.readthedocs.io/nl/latest/faq.html#recalculate-prices-retroactively docs as it seems to show an invalid command).

It took a while to delete all stats and consumptions and to set the reading table processed field back to False, but after that the backend process started to refill everything nicely.

After 2 days all readings were processed, but I noticed that the stats and consumption nodes were not completely filled. Only 1 month at the beginning and the last couple of days were added in the consumption and stats tables.

I have tried find a way to update it manually, after turing of the dsmr_backend process, but have not found anything. Maybe you could give some advice.

Thanks, Koen

dennissiemensma, 2017-08-20:
Bedankt voor je ticket. Ik heb de typo in de docs aangepast, zodat die bij een volgende release goed staat, bedankt voor het melden daarvan.

Wat betreft het hergenereren van de gegevens, het lijkt erop dat dit (nog) niet goed werkt in combinatie met de datalogger aan. 

Kan het zijn dat die nog aanstond tijdens het hergenereren? Ik heb namelijk slechts 1x eerder gehad dat iemand dit nodig had, en [die gaf aan dat de datalogger in de weg zat](https://github.com/dennissiemensma/dsmr-reader/issues/328#issuecomment-307013093).

kvolleberg, 2017-08-20:
Hoi Dennis, de datalogger stond inderdaad nog aan. Als ik deze uitzet mis
ik natuurlijk de nieuwe data.
Ik ga vanavond weer wat dingetjes proberen. Benieuwd of jij nog ideeen hebt.
Groeten koen

dennissiemensma, 2017-08-20:
Ja klopt dat is een groot nadeel. Ik heb tot zover helaas nog niet de oorzaak kunnen vinden.

kvolleberg, 2017-08-23:
Ik heb de boel weten te repareren.

Wat ik gedaan heb, is dsmr_backend uitgezet, zodat er geen interferentie was tussen mijn processen en de achtergrondprocessen. Ondertussen heb ik wel nieuwe readings binnengekregen, dus m'n metingen zijn niet onderbroken.

Ook heb ik de tabellen met statistieken (uur/dag) en consumption (elec/gas) leeggegooid; daarnaast in dsmrreading tabel proccessed = False gezet. Hiervoor heb ik even direct in de postgresql database gehakt, dit ging sneller dan via manage.py.

Daarna heb ik dsmr_consumption.services tijdelijk herschreven om zo snel mogelijk de 2.5miljoen readings om te zetten naar Consumptions.

- ipv unprocessed()[0:1024] ==> all().iterator()
- in deze for-loop een with transaction.atomic() gebouwd met na elke 5000 metingen een flush te doen ipv na elke compact
- op regel 119 dsmr_version hardcoded gemaakt, want elke keer uitzoeken duurt langer
- op regel 56 geen get_or_create maar create gedaan; gaat sneller, maar werkt alleen als je gegarandeerd nog geen consumption hebt met betreffende timestamp
- grouping_type op regel 27 hardcoded (ook performance)

Heb dit met `manage.py dsmr_backend --run-once` gedraaid/
Dit maakte dat 500 readings in 10sec verwerkt werden, dus na elke 100sec een flush. Daarmee duurde het een uur of 15 voordat ik helemaal rond was. De metingen na start van de dsmr_backend kwamen er niet in, maar worden later wel opgevangen.

Nu heb ik dus weer een complete dataset. Voordat ik de backend weer startte in supervisor, heb ik eerst even weer je 1.8.2 versie terug gezet.

Misschien dat dit je helpt met het vinden van de oorzaak.

dennissiemensma, 2017-08-23:
Bedankt voor je uitgebreide toelichting. Mooi dat je er zelf zo snel bent uitgekomen! Wellicht dat de bug dan zit in de overlap tussen de metingen op 'onverwerkt' zetten en het hergenereren van de data, al zou dat (voor Postgres) niet uit moeten maken, gezien die versies maakt van de data.

Wat betreft het zelf behandelen van de transacties, dat is nog een hele goede performance-winst voor deze bulk actie. Ik was vergeten hoeveel dat (op grote aantallen) vertraagt.

Ik zal kijken of ik hier nog wat optimalisatie uit kan halen.

dennissiemensma, 2018-01-17:
Ik heb hier vorige maand nog even kort naar gekeken, maar zal niet zo 1-2-3 verbeteringen zonder grote wijzigingen te doen. Gezien het voor de rest wel oke draait, laat ik dit voor wat het is. Bedankt voor al je feedback in het ticket. Voor toekomstige gevallen is dat zeker nuttig. :]

closed 2018-01-17 by dennissiemensma