dsmrreader/dsmr-reader #350: dsmr 5 met api v1 leid tot overlappende telegrams

tiki1980, 2017-08-20:
Ik heb dsmr 5 meter (sagemcom xs210). Deze laat ik via een raspberry PI 1 uitlezen en naar mijn server sturen via de API calls. Nu heb ik daar op 1,2% van de berichten een CRC error. Als ik hierbij het bericht zoek zie ik dat het telegram overlapped met het volgende telegram. 

In de API v1 code checken we wel op een startswith van / maar niet als die midden in het bericht staat. Dit zouden we kunnen oplossen met een if data.find('/') <> -1:  en dan de karakters voor de / van die string wissen zodat het tweede bericht toch goed verwerkt wordt. Maar dan wordt het eerste bericht niet meer verwerkt.

Ik heb helaas geen raspberry pi 3 om te testen of die dit probleem ook heeft maar kan me voorstellen dat de 1x per seconde processen van de berichten hier ook een probleem is.

**Log op de server: let op de base 64 is aangepast zodat het serie nummer niet leesbaar is**
`[2017-08-20 08:43:08,409] WARNING @ services | Rejected telegram (base64 encoded): b'L0VuZTVcWFMyMTAgRVNNUiA1LjANCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDE3MDgyMDEwNDI1OVMpDQowLTA6OTYuMS4xKHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHgpDQoxLTA6MS44LjEoMDAwMDE4LjYyMCprV2gpDQoxLTA6MS44LjIoMDAwMDAzLjcwMyprV2gpDQoxLTA6Mi44LjEoMDAwMDAwLjA3MiprV2gpDQoxLTA6Mi44LjIoMDAwMDAwLjAwMCprV2gpDQowLTA6OTYuMTQuMCgwMDAxKQ0KMS0wOjEuNy4wKDAwLjM3NiprVykNCjEtMDoyLjcuMCgwMC4wMDAqa1cpDQowLTA6OTYuNy4yMSgwMDAwNCkNCjAtMDo5Ni43LjkoMDAwMDEpDQoxLTA6OTkuOTcuMCgwKSgwLTA6OTYuNy4xOSkNCjEtMDozMi4zMi4wKDAwMDAyKQ0KMS0wOjMyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6L0VuZTVcWFMyMTAgRVNNUiA1LjANCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDE3MDgyMDEwNDMwMVMpDQowLTA6OTYuMS4xKHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHgpDQoxLTA6MS44LjEoMDAwMDE4LjYyMSprV2gpDQoxLTA6MS44LjIoMDAwMDAzLjcwMyprV2gpDQoxLTA6Mi44LjEoMDAwMDAwLjA3MiprV2gpDQoxLTA6Mi44LjIoMDAwMDAwLjAwMCprV2gpDQowLTA6OTYuMTQuMCgwMDAxKQ0KMS0wOjEuNy4wKDAwLjM3NyprVykNCjEtMDoyLjcuMCgwMC4wMDAqa1cpDQowLTA6OTYuNy4yMSgwMDAwNCkNCjAtMDo5Ni43LjkoMDAwMDEpDQoxLTA6OTkuOTcuMCgwKSgwLTA6OTYuNy4xOSkNCjEtMDozMi4zMi4wKDAwMDAyKQ0KMS0wOjMyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIyOS4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjIxLjcuMCgwMC4zNzcqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjAtMToyNC4xLjAoMDAzKQ0KMC0xOjk2LjEuMCh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4KQ0KMC0xOjI0LjIuMSgxNzA4MjAxMDQwMDBTKSgwMDAwMS4xOTAqbTMpDQohMDA4RQ=='
[2017-08-20 08:43:08,410] ERROR @ services | CRC mismatch: 142 (telegram) != 55067 (calculated)
Traceback (most recent call last):
  File "/root/dsmr-reader/dsmr_datalogger/services.py", line 179, in telegram_to_reading
    verify_telegram_checksum(data=data)
  File "/root/dsmr-reader/dsmr_datalogger/services.py", line 131, in verify_telegram_checksum
    'CRC mismatch: {} (telegram) != {} (calculated)'.format(telegram_checksum, calculated_checksum)
dsmr_datalogger.exceptions.InvalidTelegramError: CRC mismatch: 142 (telegram) (calculated)
`

**bericht op de raspberry pi1**
`/Ene5\XS210 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(170820104259S)
0-0:96.1.1(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
1-0:1.8.1(000018.620*kWh)
1-0:1.8.2(000003.703*kWh)
1-0:2.8.1(000000.072*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.376*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00004)
0-0:96.7.9(00001)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00002)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:/Ene5\XS210 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(170820104301S)
0-0:96.1.1(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
1-0:1.8.1(000018.621*kWh)
1-0:1.8.2(000003.703*kWh)
1-0:2.8.1(000000.072*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.377*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00004)
0-0:96.7.9(00001)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00002)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(229.0*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.377*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
0-1:24.2.1(170820104000S)(00001.190*m3)
!008E`

dennissiemensma, 2017-08-20:
Bedankt voor je melding. Ik heb hier helaas geen concrete oplossing voor. De applicatie leest simpelweg wat die binnenkrijgt en wanneer berichten overschreven worden, dan kan deze er weinig mee doen. 

Ik heb zelf overigens geen DSMR v5 meter, dus ik heb dit alleen teruggezien in feedback van anderen (zoals die van jou hierboven). Als er iets te verbeteren valt, dan zou dat in het dataproces moeten zitten wat de kabel uitleest. Ik weet bijvoorbeeld niet of het een issue is in de code of een technisch probleem als gevolg van de DSMR v5 telegram snelheid (elke seconde).

Je zou nog kunnen proberen om een andere client te gebruiken, zoals `ndokter/dsmr_parser`. Alleen zul je dan wel de v2 API moeten gebruiken.

tiki1980, 2017-08-22:
Nog wat zaken getest maar nog geen oplossing:

- Raspberry PI 3 geeft hetzelfde probleem met de api
- Raspberry PI 3 geeft hetzelfde probleem met de software lokaal geinstalleerd (behalve de postgresql database)
- Als ik met `cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q` de output opsla na file komt het probleem ook voor dus het is geen specifiek dsmr-reader probleem.

wat ik nog kan bedenken:
- kabel niet goed / storing
- linux issue met near realtime data op de seriele poort (buffer etc, al zie ik geen overrun fouten in de logs)
- issue in de meter software....

dennissiemensma, 2017-08-23:
Wellicht is het ook niet op te lossen. En voor een DSMR v5 meter vind ik 1,2% aan verloren metingen ook weer niet superveel (wel qua absolute getallen overigens). 
Met mijn DSMR v4 meter heb ik max 6 metingen per minuut, in jouw geval zit je zelfs met het verlies op nog zo'n 58 per minuut. Plus dat je de nauwkeurigheid van het gasverbruik (elke 5 a 10 minuten?) behoudt.

Niet om goed te praten dat dit gebeurt, maar wel om het even in perspectief te plaatsen.

tiki1980, 2017-08-24:
Aangezien het geen dsmr-reader probleem is zal ik het ticket sluiten, mocht ik nog een oplossing vinden dan laat ik het weten.  (ik kan het natuurlijk niet uitstaan dat ik niet weet waardoor dit veroorzaakt wordt). 

De feature om metingen te archiveren wordt overigens  met een dsmr 5 wel belangrijker. (1 jaar aan metingen is 86.400 per dag en 31.536.000 per jaar). Of misschien zelfs een feature om er maar x per minuut op te slaan.

closed 2017-08-24 by tiki1980

dennissiemensma, 2017-08-24:
Archivering (of liever gezegd, retentie) staat in #72. In de zomermaanden ontwikkel ik doorgaans niet actief, dus in het najaar ga ik weer verder. Te beginnen met #9.