dsmrreader/dsmr-reader #352: Inconsistent implementation 1.7.0 with Sagemcom T210-D

Herugrim, 2017-08-28:


dennissiemensma, 2017-08-28:
Hi, thank you for reporting this issue. I'm not really sure what the differences are, could you provide some additional info? Both the  `1.7.0`  field and the phase fields should be set by the meters.

Do you mean that the E350 does not set the `1.7.0` value? Because as far as I know, it should work properly, but maybe I'm misinterpreting something here. :]

Herugrim, 2017-08-28:
As you can see in the example, the E350 sets `1.7.0` to `0` even though `41.7.0` is `054` as it adds the values of `22.7.0`, `42.7.0` and `62.7.0`, deducts the values of `21.7.0`, `41.7.0` and `61.7.0` and sets this value for `1.7.0`. 

The T210-D example shows that this meter  adds the values of `22.7.0`, `42.7.0` and `62.7.0` and sets this as `2.7.0`. It also adds the values of `21.7.0`, `41.7.0` and `61.7.0` and sets this value for `1.7.0`.  I had a short look in the spec but i could not find how `1.7.0` 'should' be calculated.

As a result DSMR-reader will show both delivery and return energy in the dashboard at the same time with the T210-D and this confused me. The following picture of the T210-D dashboard would never occur with the E350 (that would give either a flat red or green line), see http://splur.nl/t210-d.png  . I suspect this might confuse other users and thus the FYI. 

I do not know what the other 3 phase meters is, perhaps you some telegrams that would show this.

dennissiemensma, 2017-08-29:
Thank you for clarifying a bit. I don't have solar panels myself, so I'm not sure how it should work. But I guess that it's possible that the smart meter reports both electricity consumed & returned at the same time (that's what you are pointing at, right?).

I think any electricity returned (to the phase it's connected to), will only reduce/zero the 'net' electricity consumed of that single phase. But I'm really not sure, since I do not have solar panels myself (I mock all such data).

Maybe you can ask some other solar panel owners on (other) forums for their telegrams to compare with. The application reports what it sees, so I don't think I can do much about that to be honest. I appreciate the FYI though, for any future questions.

dennissiemensma, 2017-09-18:
I will close this for now, as I do not have a fix for this.

closed 2017-09-18 by dennissiemensma

Greetingandgreat, 2017-11-24:
hi guys, maybe I can ask someone for this issue

dennissiemensma, 2017-11-24:
What is your question? You can also create a new issue if you want to know something (unrelated).