dsmrreader/dsmr-reader #354: API V2 Telegram parser example

aoturoa, 2017-09-06:
First of all, great project and very nicely written code.
The REST API is an awesome amendment to the project and opens up lots of possibilities.

What I am missing though in the REST API docs is a _Telegram parser example_ to post a JSON object. API v1 example is very complete, but for REST API there is a gap ... how to go from raw telegram to JSON object. 

It seems all the logic is available in _dsmr_datalogger_.
Can that somehow be used for a client file as in your example v1 'dsmr_datalogger_api_client.py' ?

dennissiemensma, 2017-09-07:
Well, I guess you can scrap some code of the datalogger app, but the point of the v2 API is that you can use any client side parser of your choice. Such [as this one](https://github.com/ndokter/dsmr_parser).

The v1 API can be used if you don't want to bother with parsing the data.

aoturoa, 2017-09-08:
You are right. If the v1 and v2 API's both are supported there is no need to parse the raw telegram before sending it to v2 API. Maybe it's an idea to add the suggested parser in the docs?
Thank you for quick response. Keep up the good work.

closed 2017-09-08 by aoturoa