dsmrreader/dsmr-reader #355: Postgress still busy after +- 300.000 telegrams

joopy, 2017-09-10:
There where ~300.000 unprocessed telegrams (~ 4 weeks). After raspberry pi 2 reboot processing started. After ~ 35 hours al telegrams where processed.

Still slow Raspberry, htop reveals postgress is still doing magic. So for more dan 3 day's I have a almost 90% cpu with postgress.

psql statistics:
postgres-# order by pg_database_size(datname) desc;
  datname   | pg_size_pretty
------------+----------------
 dsmrreader | 499 MB
 postgres   | 6516 kB
 template1  | 6417 kB
 template0  | 6409 kB
(4 rows)

984 | -00:00:00.782917 | dsmrreader | COMMIT
 1036 | -00:00:17.485937 | dsmrreader | SELECT "dsmr_datalogger_dsmrreading"."id", "dsmr_datalogger_dsmrreading"."timestamp", "dsmr_datalogger_dsmrreading"."electricity_delivered_1", "dsmr_datalogger_dsmrreading"."electricity_returned_1", "dsmr_datalogger_dsmrreading"."electricity_delivered_2", "dsmr_datalogger_dsmrreading"."electricity_returned_2", "dsmr_datalogger_dsmrreading"."electricity_currently_delivered", "dsmr_datalogger_dsmrreading"."electricity_currently_returned", "dsmr_datalogger_dsmrreading"."phase_currently_delivered_l1", "dsmr_datalogger_dsmrreading"."phase_currently_delivered_l2", "dsmr_datalogger_dsmrreading"."phase_currently_delivered_l3", "dsmr_datalogger_dsmrreading"."extra_device_timestamp", "dsmr_datalogger_dsmrreading"."extra_device_delivered", "dsmr_datalogger_dsmrreading"."processed" FROM "dsmr_datalogger_dsmrreading" WHERE "dsmr_datalogger_dsmrreading"."processed" = false ORDER BY "dsmr_datalogger_dsmrreading"."timestamp" ASC LIMIT 1024

SELECT max(now() - xact_start) FROM pg_stat_activity WHERE state IN ('idle in transaction', 'active');
       max
-----------------
 00:00:01.036794
(1 row)

Any Idea's? Running second to last DSMR version

dennissiemensma, 2017-09-11:
I asume you have tried restartingall  the supervisor processes? Or the entire system?

joopy, 2017-09-12:
Ja de raspberry heeft meerdere keren een reboot gekregen denk 2 keer. Maar het is nog steeds erg druk zo druk dat de webserver het niet trekt. Enig idee dennis?

dennissiemensma, 2017-09-12:
- Welke processen zie je dan als je `SELECT pid, datname, usename, query_start, query FROM pg_stat_activity;` doet?

- En als je op de command line dit doet, wat staat daar allemaal bij? `sudo ps faux | grep postg`

- Mocht je daar nog niets bijzonders in zien, wellicht kun je het zien aan de systemcalls die het betreffende (90% cpu) proces doet met: `strace -p PID` (PID = process ID)

joopy, 2017-09-15:
Systeem krijgt iets meer lucht. Heb de pingdom check extern eraf gehaald. Niet elke x min de site opvragen.

dennissiemensma, 2017-09-15:
Maar verder dus niet gezien welke queries of processen er uitgevoerd worden?

joopy, 2017-09-15:
Thanks voor de hulp. Zonder elke min een pingdom check trekt de Raspberry het prima.

* Welke processen zie je dan als je SELECT pid, datname, usename, query_start, query FROM pg_stat_activity; doet?

 pid  |  datname   |  usename   |          query_start          |                                                                                                                                                                                                                                                                                                                                                                                                                                                                  query                                                                                                                                                                                                                                                                                                                                                                                                                                      
------+------------+------------+-------------------------------+------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 3653 | dsmrreader | dsmrreader | 2017-09-15 20:04:16.138601+00 | SELECT "dsmr_datalogger_dataloggersettings"."id", "dsmr_datalogger_dataloggersettings"."track", "dsmr_datalogger_dataloggersettings"."track_phases", "dsmr_datalogger_dataloggersettings"."verify_telegram_crc", "dsmr_datalogger_dataloggersettings"."dsmr_version", "dsmr_datalogger_dataloggersettings"."com_port" FROM "dsmr_datalogger_dataloggersettings" WHERE "dsmr_datalogger_dataloggersettings"."id" = 1
 3658 | dsmrreader | dsmrreader | 2017-09-15 20:04:22.064633+00 | SELECT "dsmr_consumption_electricityconsumption"."id", "dsmr_consumption_electricityconsumption"."read_at", "dsmr_consumption_electricityconsumption"."delivered_1", "dsmr_consumption_electricityconsumption"."returned_1", "dsmr_consumption_electricityconsumption"."delivered_2", "dsmr_consumption_electricityconsumption"."returned_2", "dsmr_consumption_electricityconsumption"."currently_delivered", "dsmr_consumption_electricityconsumption"."currently_returned", "dsmr_consumption_electricityconsumption"."phase_currently_delivered_l1", "dsmr_consumption_electricityconsumption"."phase_currently_delivered_l2", "dsmr_consumption_electricityconsumption"."phase_currently_delivered_l3" FROM "dsmr_consumption_electricityconsumption" WHERE "dsmr_consumption_electricityconsumption"."read_at" > '2017-09-15T00:00:00+02:00'::timestamptz ORDER BY "dsmr_consumption_electricityconsumption"."read_at" ASC LIMIT 1
 3683 | dsmrreader | dsmrreader | 2017-09-15 20:02:13.882119+00 | SELECT "dsmr_datalogger_meterstatistics"."id", "dsmr_datalogger_meterstatistics"."timestamp", "dsmr_datalogger_meterstatistics"."dsmr_version", "dsmr_datalogger_meterstatistics"."electricity_tariff", "dsmr_datalogger_meterstatistics"."power_failure_count", "dsmr_datalogger_meterstatistics"."long_power_failure_count", "dsmr_datalogger_meterstatistics"."voltage_sag_count_l1", "dsmr_datalogger_meterstatistics"."voltage_sag_count_l2", "dsmr_datalogger_meterstatistics"."voltage_sag_count_l3", "dsmr_datalogger_meterstatistics"."voltage_swell_count_l1", "dsmr_datalogger_meterstatistics"."voltage_swell_count_l2", "dsmr_datalogger_meterstatistics"."voltage_swell_count_l3", "dsmr_datalogger_meterstatistics"."rejected_telegrams" FROM "dsmr_datalogger_meterstatistics" WHERE "dsmr_datalogger_meterstatistics"."id" = 1
 3735 | postgres   | postgres   | 2017-09-15 20:04:21.994661+00 | SELECT pid, datname, usename, query_start, query FROM pg_stat_activity;
(4 rows)

root@raspberrypi:~# sudo ps faux | grep postg
root      3764  0.0  0.4   4264  2012 pts/1    S+   20:05   0:00                          \_ grep postg
postgres   657  0.0  4.0 161044 18056 ?        S    18:48   0:00 /usr/lib/postgresql/9.4/bin/postgres -D /var/lib/postgresql/9.4/main -c config_file=/etc/postgresql/9.4/main/postgresql.conf
postgres   811  0.0  2.0 161144  8948 ?        Ss   18:48   0:00  \_ postgres: checkpointer process
postgres   812  0.0  1.1 161044  5060 ?        Ss   18:48   0:00  \_ postgres: writer process
postgres   813  0.0  1.8 161044  8284 ?        Ss   18:48   0:00  \_ postgres: wal writer process
postgres   814  0.0  1.2 161328  5696 ?        Ss   18:48   0:00  \_ postgres: autovacuum launcher process
postgres   815  0.0  0.8  18564  3868 ?        Ss   18:48   0:02  \_ postgres: stats collector process
[... cut ...]
poll([{fd=7, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=7, revents=POLLIN}])
rt_sigprocmask(SIG_BLOCK, [PIPE], [], 8) = 0
read(7, "\27\3\3\1+", 5)                = 5
read(7, "\316\276\24[\265\225\v\202\315\377G\337\320\273'\256\244\235[O\327\33\377\306\214\345\244\331N\306\206\255"..., 299) = 299
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
gettimeofday({1505505973, 195395}, NULL) = 0
write(1, "Command completed. Sleeping for "..., 47) = 47
write(1, "\n", 1)                       = 1
select(0, NULL, NULL, NULL, {1, 0}^CProcess 3641 detached
 <detached ...>

closed 2017-09-15 by joopy

dennissiemensma, 2017-09-15:
Oke, wellicht was het dan iets te veel met dat pingdom erbij. Ik zie geen vreemde zaken als ik even snel een blik werp.