dsmrreader/dsmr-reader #356: Update to 1.8.2 not working?

joopy, 2017-09-15:
(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
1.8.1

 Please make sure you've read EACH release note BEFORE deploying, since some
 changes might not always be backwards compatible or depend on your database!

     >>>   https://github.com/dennissiemensma/dsmr-reader/releases   <<<

 ~~~ Still want to deploy? (type Y + ENTER to continue or press CTRL + C to abor                                                                                                                                                                                                                                          t)
y

 --- Pulling remote repository for new commits...
**./deploy.sh: line 18: git: command not found**

 --- Merging/updating checkout.
**./deploy.sh: line 24: git: command not found**

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking & synchronizing base requirements for changes.
Requirement already satisfied (use --upgrade to upgrade): crcmod==1.7 in /home/d                                                                                                                                                                                                                                          smr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader/prov                                                                                                                                                                                                                                          isioning/requirements/base.txt (line 1))
Requirement already satisfied (use --upgrade to upgrade): coreapi==2.3.1 in /hom                                                                                                                                                                                                                                          e/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader/p                                                                                                                                                                                                                                          rovisioning/requirements/base.txt (line 2))
Requirement already satisfied (use --upgrade to upgrade): django==1.10.7 in /hom                                                                                                                                                                                                                                          e/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader/p                                                                                                                                                                                                                                          rovisioning/requirements/base.txt (line 3))
Requirement already satisfied (use --upgrade to upgrade): django-colorfield==0.1                                                                                                                                                                                                                                          .14 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r d                                                                                                                                                                                                                                          smrreader/provisioning/requirements/base.txt (line 4))
Requirement already satisfied (use --upgrade to upgrade): django-debug-toolbar==                                                                                                                                                                                                                                          1.8 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r d                                                                                                                                                                                                                                          smrreader/provisioning/requirements/base.txt (line 5))
Requirement already satisfied (use --upgrade to upgrade): django-filter==1.0.4 i                                                                                                                                                                                                                                          n /home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrre                                                                                                                                                                                                                                          ader/provisioning/requirements/base.txt (line 6))
Requirement already satisfied (use --upgrade to upgrade): djangorestframework==3                                                                                                                                                                                                                                          .6.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r                                                                                                                                                                                                                                           dsmrreader/provisioning/requirements/base.txt (line 7))
Requirement already satisfied (use --upgrade to upgrade): django-solo==1.1.2 in                                                                                                                                                                                                                                           /home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrread                                                                                                                                                                                                                                          er/provisioning/requirements/base.txt (line 8))
Requirement already satisfied (use --upgrade to upgrade): dropbox==8.0.0 in /hom                                                                                                                                                                                                                                          e/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader/p                                                                                                                                                                                                                                          rovisioning/requirements/base.txt (line 9))
Requirement already satisfied (use --upgrade to upgrade): gunicorn==19.7.1 in /h                                                                                                                                                                                                                                          ome/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader                                                                                                                                                                                                                                          /provisioning/requirements/base.txt (line 10))
Requirement already satisfied (use --upgrade to upgrade): paho-mqtt==1.2.3 in /h                                                                                                                                                                                                                                          ome/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages (from -r dsmrreader                                                                                                                                                                                                                                          /provisioning/requirements/base.txt (line 11))
[... cut ...] ation, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 363 unmodified.

 --- Reloading app code...

 * Reloading Gunicorn (webinterface)...
 [x] Done!

 * Reloading backend process...
 [x] Done!

 * Reloading datalogger process...
 [x] Done!

 --- Clearing cache...
Clearing cache: default
Done

 --- Deployed version:
**1.8.1**

 >>> Deployment complete. <<<

dennissiemensma, 2017-09-15:
GIT lijkt niet ingestalleerd: `git: command not found`

Al geprobeerd als sudo-user: `sudo apt-get install git` ?

joopy, 2017-09-15:
facepalm.. Ik had de deploy.sh al met nano bekeken maar geen belletje ik zag alleen een echo staan. Ik had pihole op mijn raspberry pi. Maar die is niet verhuist naar de rpi 3 met home assistant. Dank Dennis.

closed 2017-09-15 by joopy