dsmrreader/dsmr-reader #357: Error Statistieken pagina bij meerdere leveranciers

Bl4ckviper, 2017-09-17:
In mijn instellingen heb ik netjes de leverancier van gas en elektra (deze zijn verschillend) apart genoemd. 

Alleen lijk het dat de statistieken pagina hier niet mee om kan gaan die geeft namelijk:

Exception:

MultipleObjectsReturned: get() returned more than one EnergySupplierPrice -- it returned 2!
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/handlers/base.py", line 185, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 68, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 88, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 155, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/statistics.py", line 33, in get_context_data
    context_data['energy_prices'] = EnergySupplierPrice.objects.by_date(today)

  File "/home/dsmr/dsmr-reader/dsmr_consumption/models/energysupplier.py", line 12, in by_date
    start__lte=target_date,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 389, in get
    (self.model._meta.object_name, num)

Bl4ckviper, 2017-09-17:
ok ;) kwam er pas vandaag achter omdat het andere contract inging.

dennissiemensma, 2018-06-11:
Ik heb hier nog even naar gekeken voor een fatsoenlijke fix, maar ik kan helaas geen goede oplossing vinden. Het combineren van contracten, en vooral de verschillende prijzen, is wat lastig. Helemaal wanneer iemand dezelfde prijsveld voor verschillende contracten in dezelfde periode invult (de foutafhandeling). 

Ik denk dat je helaas toch aangewezen bent op het combineren van de prijzen in één contract.

closed 2018-06-11 by dennissiemensma

Bl4ckviper, 2018-06-11:
Ok, ik heb inmiddels toch geen gas meer dus ook geen probleem meer ;) 

Bedankt voor de update!