dsmrreader/dsmr-reader #359: Te veel onverwerkte telegrammen

basvdburg, 2017-09-19:
Hallo,

Ik gebruik sinds een aantal dagen DSMR reader (waar ik tot nu toe heel tevreden mee ben!). En mij is opgevallen dat het aantal onverwerkte telegrammen per minuut erg snel op loopt. Ik heb een Landis+Gyr E350 SMR 5.0 die iedere seconde een bericht via de seriële ingang terug geeft. En mijn vermoeden is dat de Raspberry Pi 3 dat niet snel genoeg kan verwerken. In veel gevallen loopt de waarde op richting de 60, en reset deze weer. Heb in de interface van DSMR reader aangegeven dat hij de berichten per minuut moet groeperen. Maar kan het kwaad dat deze waarde zo hoog op loopt?

nheinemans, 2017-09-19:
Ik heb DSMR in een docker container op een RPi 3b draaien, en mijn meter geeft ook elke seconde een bericht. De teller loopt ook op, ik heb niet bijgehouden hoe hoog (toen ik net keek was de maximale waarde 37). Zo te zien komt uiteindelijk alle info in de database en kloppen de grafieken.

dennissiemensma, 2017-09-19:
Bedankt voor jullie vragen. De werking is zoals bedoeld, mits jullie inderdaad beiden de optie `groeperen per minuut` aan hebben staan. 
De applicatie 'spaart' net zo lang telegrammen op totdat de minuut voorbij is, zodat ze verwerkt en gegroepeerd kunnen worden. Als de teller ongeveer elke minuut gereset wordt, dan kan de rPi het makkelijk aan hoor. :]

Er zou ook nog wat extra uitleg moeten staan op de Status-pagina, wanneer je klikt op `Klik hier voor meer informatie`. Mochten jullie nog meer gekke dingen tegenkomen, laat het vooral weten.

closed 2017-09-19 by dennissiemensma