dsmrreader/dsmr-reader #360: Gestapelde staafdiagramen in Archief en Vergelijk

vgouw, 2017-09-30:
Issue
In "Archief" en "Vergelijk" worden  lijn-grafieken gebruikt.  ik ben van mening dat:
- De lijn-grafiek onterecht een verloop aangeeft, terwijl het absolute/totaal uur/dag/maand waardes zijn 
- Voor elektriciteit is elk tarief apart getoond. Hierdoor is het moeilijk om weekend-dagen (alleen laag tarief) te vergelijken met niet weekend-dagen (normaal en laag tarief)

Voorstel
Gebruik gestapelde staaf diagrammen voor "Archief" en "Vergelijk"

dennissiemensma, 2017-10-01:
Bedankt voor je input! Mogelijk neem ik het mee voor #236, wanneer ik daar aan toekom. Deze staat sowieso niet voor de korte termijn op de agenda.

dennissiemensma, 2018-03-05:
Zoals in gerelateerde tickets: redesign zie ik niet zo 1-2-3 gebeuren voorlopig.

closed 2018-03-05 by dennissiemensma