dsmrreader/dsmr-reader #361: ISKRA AM550-ED1.01 werkt niet

kuhlivisj, 2017-10-06:
Er is bij mij vanmiddag een slimme meter geplaatst.
Kabel er op aangesloten en correct ingesteld in DSMR reader.
De telegrammen komen netjes binnen, maar DSMR reader accepteert ze niet.
Als ik in django.log kijk zie ik daar steeds de volgende melding:

`
[2017-10-06 15:16:40,332] ERROR @ services | Discarded telegram with future timestamp: /ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(171006151639S)
0-0:96.1.1(4530303433303036393234323439323137)
1-0:1.8.1(000000.558*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.343*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(170712072144S)(0000000565*s)
1-0:32.32.0(00005)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(227.5*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.345*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(700101010000W)(00000000)
!5440
`

Wat gaat er hier fout?

dennissiemensma, 2017-10-06:
Bedankt voor het melden. Het lijkt erop dat er nog een bug zit in de verwerking van het gas-tijdstip in dit geval. Ik heb je telegram hier even geanalyseerd en de applicatie maakt er dit van:

```
{'dsmr_version': '50',
 'electricity_currently_delivered': '00.343kW',
 'electricity_currently_returned': '00.000kW',
 'electricity_delivered_1': '000000.558kWh',
 'electricity_delivered_2': '000000.000kWh',
 'electricity_returned_1': '000000.000kWh',
 'electricity_returned_2': '000000.000kWh',
 'electricity_tariff': '0001',
 'extra_device_delivered': '00000000',
 'extra_device_timestamp': datetime.datetime(2070, 1, 1, 0, 0, tzinfo=<UTC>),
 'long_power_failure_count': '00003',
 'phase_currently_delivered_l1': '00.345kW',
 'phase_currently_delivered_l2': None,
 'phase_currently_delivered_l3': None,
 'power_failure_count': '00008',
 'timestamp': datetime.datetime(2017, 10, 6, 13, 16, 39, tzinfo=<UTC>),
 'voltage_sag_count_l1': '00005',
 'voltage_sag_count_l2': None,
 'voltage_sag_count_l3': None,
 'voltage_swell_count_l1': '00001',
 'voltage_swell_count_l2': None,
 'voltage_swell_count_l3': None}
```

Het gaat fout bij het analyseren van dit deel:
```
0-1:24.2.1(700101010000W)(00000000)
```

De applicatie maakt er dit van:
```
 'extra_device_timestamp': datetime.datetime(2070, 1, 1, 0, 0, tzinfo=<UTC>),
```

En doordat het tijdstip (ver) in de toekomst ligt, `2070`, negeert de applicatie dit. Die beperking is ingebouwd vanwege #269. Alleen kwam het daar **incidenteel** voor en bij jou lijkt het constant.

Klopt het dat je **geen** gasmeter hebt die aangesloten is op de slimme meter?

kuhlivisj, 2017-10-06:
Ik heb wel een gasmeter. Het kan alleen zijn dat die nog niet gecommuniceerd heeft omdat hij pas 2 uurtjes hangt en de stand nog nagenoeg op 0 staat.

dennissiemensma, 2017-10-06:
Ik denk dat je dan het beste even (een dagje) kunt wachten of de situatie verandert. Momenteel geeft ofwel je gasmeter constant `2070` terug, ofwel je slimme meter werkt de datum niet bij.

Heeft de monteur wel aangegeven dat ze met elkaar gekoppeld zijn? Het is namelijk geen vereiste.

dennissiemensma, 2017-10-07:
@kuhlivisj hoe zijn de meterstanden nu? Is er iets veranderd en zie je nu wel je gasstanden (correct) terug?

kuhlivisj, 2017-10-07:
Hij werkt nu. Afgelopen nacht is de gasmeter de stand door gaan geven aan de elektriciteits meter. De stand van de gasmeter wordt nu netjes elke 5 minuten doorgegeven.

dennissiemensma, 2017-10-07:
Top, fijn om te horen! 

En ook goed voor mij om te weten wanneer iemand tegen een soortgelijke situatie aanloopt. Bedankt

closed 2017-10-07 by dennissiemensma