dsmrreader/dsmr-reader #364: Update v1.9 broke telegram reading (only first telegram read after boot)

PeterEIER, 2017-10-10:
All process are running. TTY tool shows that telegrams are read / comming in. DSMR reader only reads the first telegram after boot. Siggestions?

dennissiemensma, 2017-10-10:
Bedankt voor je melding. Voor zover ik zo 1-2-3 kan zien is er tussen 1.8.2 en 1.9.0 niets veranderd in de datalogger.

* Wat is de output wanneer je in Supervisor `restart dsmr_datalogger` doet? Zie je dan wel initiele output?
* En staan er nog fouten in de logfiles in `home/dsmr/dsmr-reader/logs/`?

* Mogelijk ligt het aan de update van `pyserial` van `3.3` naar `3.4`. Je kunt dit proberen:
** `sudo su - dsmr`
** `pip install pyserial==3.3`
** In supervisor (als sudo/root) `restart dsmr_datalogger` doen.

----

Je kunt eventueel helemaal terug naar de vorige versie met:
* `sudo su - dsmr`
* `./manage.py migrate dsmr_consumption 0007_settings_documentation` (deze is nodig omdat een aantal database velden hernoemd waren, als het goed is is de output: ` Unapplying dsmr_consumption.0008_energysupplier_return_prices... OK`)
* `git checkout tags/v1.8.2`
* `./deploy.sh`
* Je zit nu als het goed is weer op `v1.8.2`

----

En weer naar 1.9 met:
* `sudo su - dsmr`
* `git checkout master`
* `./deploy.sh` (voert ook de DB migratie weer uit)
* Je zit nu als het goed is weer op `v1.9.0`

PeterEIER, 2017-10-11:
Hallo Dennis,

 

Bedankt voor je snelle reactie!

 

Pyserial downgrade naar v 3.3 heeft mijn probleem opgelost! Werkt nu weer als een zonnetje, bedankt voor je hulp.

 

Met vriendelijke groet,

Peter

PeterEIER, 2017-10-11:
Was vergeten mij screenshot van de dsmrreader.log te attachen…. Laat de error zien gisteravond, en vanmorgen na de downgrade van pyserial naar 3.3. Hopelijk geeft dit een aanknoopingspunt. 

 

 

 

 

Met vriendelijke groet,

Peter

dennissiemensma, 2017-10-11:
Bedankt voor het uitproberen en de terugkoppeling. In dat geval zal ik een patch uitbrengen die pyserial weer downgrade. Ik heb verder van niemand klachten gehad dat het stuk ging, maar gezien de vorige pyserial in `v1.8.2` het prima deed, houd ik die wel aan.

PeterEIER, 2017-10-11:
Dennis,

 

Ik doe net een ontdekking… ik experimenteer wat met MQTT… MQTT staat default aan bij mij in DSMR reader…. Gisteren bij het trouble shooten MQTT uitgezet… vanmorgen de downgrade gedaan alles weer werken… Nu zet ik zojuist MQTT weer aan… het lezen van de telegrams stopt spontaan!! MQTT weer uitgezet helpt niets… nu doe ik een pyserial upgrade naar 3.4… en telegrams werken weer (met pyserial 3.4)… weer MQTT aangezet… meteen weer problemen… Een “restart dsmr_datalogger” is voldoende om het probleem te verhelpen. Mij conclusie is dat het niet aan pyserial 3.4 ligt… maar iets met MQTT te maken heeft (wellicht link met PVOUTPUT???).

 

Alleen deze MQTT switch aangezet, rest staat “disabled”, en het is meteen mis. Wanneer ik “restart dsmr_datalogger” doe werkt het weer.

 

 

 

Met vriendelijke groet,

Peter

dennissiemensma, 2017-10-11:
Bedankt voor je aanvulling. In de vorige release is MQTT-support ongewijzigd. Verder doet PVOutput alleen een export naar hun website en niets met de datalogger.

Kan het zijn dat een ander proces op je Pi ook de seriele poort probeert uit te lezen?

PeterEIER, 2017-10-11:
Op deze Pi3 draait alleen DSMR reader en verder niets.

dennissiemensma, 2017-10-11:
Hmmm. Ik heb het hier ook nog geprobeerd te reproduceren, inclusief datalogger en MQTT. Zowel de datalogger als MQTT komt gewoon binnen.

Het enige verschil is dan dat ik nog een 'trage' DSMR v4 meter heb. Misschien dat de oorzaak zit in de snelheid van je DSMR v5 meter, maar ik kan dat hier niet reproduceren.

Voor nu kan ik je alleen adviseren om dan MQTT uit te zetten, als dat het voor je oplost. Als er iets met MQTT is, dan verwacht ik nog wel dat anderen zich hier melden.

PeterEIER, 2017-10-11:
Ja, is ook geen probleem. Bedankt voor het onderzoeken/meedenken. Mocht ik zelf nog iets vinden laat ik je dat weten.

dennissiemensma, 2017-10-23:
Voor nu sluit ik deze. Tot nu toe heb ik verder niemand anders gehoord over een soortgelijk issue, helaas.

closed 2017-10-23 by dennissiemensma