dsmrreader/dsmr-reader #367: Dagverbruik en teruglevering via MQTT

Ernst79, 2017-10-20:
Allereerst bedankt voor deze geweldige applicatie! Ik heb een feature request. 

Is het wellicht mogelijk om ook het huidige dagverbruik en teruglevering via MQTT te laten versturen? Zeg maar de dagwaarden in de tabel op de dashboard pagina. Op dit moment kan ik alleen het huidige gebruik in Watt en de meterstanden laten versturen, maar ik zou graag het dagverbruik in KWh ook willen. Dan kan ik die weer intrigeren in home assistant.

dennissiemensma, 2017-10-21:
Bedankt voor je suggestie, dat lijkt me een prima toevoeging. 

Doel je op het dagverbruik dat 1x per dag gemaakt wordt? Of op een tussenstand die elke X interval doorgestuurd wordt?

Ernst79, 2017-10-21:
Ik doel op de tussenstand die elke X interval wordt doorgestuurd, oftewel het verbruik van vandaag wat tot nu toe is verbruikt/opgewekt. Dus dat zal gedurende de dag oplopen (en midden in de nacht weer op nul beginnen). 

Dit zijn dus de getallen die je in de bovenste tabel op het dashboard ook toont. 

Bedankt in ieder geval!

dennissiemensma, 2017-10-22:
Oke bedankt voor de opheldering.

joopy, 2017-11-28:
Inderdaad! Dit zou een mooie toevoeging zijn Dennis en Ernst.

jorisvervuurt, 2017-12-05:
Om het misschien wat globaler aan te pakken: wellicht interessant om een soort plugin-structuur vast te leggen, zodat op die manier door derde partijen uitbreidingen gemaakt kunnen worden?

dennissiemensma, 2017-12-05:
Bedankt voor je suggestie. Op dit moment is dat denk ik wat veel werk qua structuur en architectuur. Voor nu voeg ik dit soort uitbreidingen sowieso wel toe, zolang er een goede use-case voor is.

jorisvervuurt, 2017-12-05:
👍 't Is dat ik meer in de hoek van Node.js/JavaScrip/PHP/C# zit; anders had ik wellicht een steentje bij kunnen dragen aan het Python-gebeuren. 😃

FutureCow, 2018-02-27:
Dit zou een fijne toevoeging zijn!

f0nt4, 2018-03-03:
Goeie suggestie dit. Ook een mqtt topic tbv het huidige tarief zou fijn zijn.

dennissiemensma, 2018-03-05:
Mijn vraag aan jullie is, is dat afdoende?

f0nt4, 2018-03-05:
@dennissiemensma Hey Dennis, thanks! Is het teveel gevraagd om ook het tarief (laag/hoog) als topic mee te kunnen sturen?

dennissiemensma, 2018-03-05:
Bedankt voor je reactie. Hoe bedoel je als topic? Als losse call per veld? 

Ik zou nog een variant kunnen maken waarbij je elk veld uit bovenstaand voorbeeld naar een apart topic kan sturen. Eigenlijk net zoals de andere MQTT-variant die ik ondersteun voor telegrammen.

Ernst79, 2018-03-05:
Lijkt mij prima, voor wat betreft de dagtotalen. Voor wat betreft de aparte topics Mqtt, dat zou wel mijn voorkeur hebben. Dat maakt het uitlezen wat makkelijker.

f0nt4, 2018-03-05:
@dennissiemensma Ik maak gebruik van de gesplitste topic telegram configuratie om bepaalde zaken weer te geven in Home Assistant. Met de huidige mogelijkheden kan ik daarbij niet simpelweg weergeven of het op het moment het lage of hoge tarief is. In de raw informatie is dit wel beschikbaar: 0-0:96.14.0: electricity_tariff.
En het idee is om dat het volgende toe te voegen aan de gesplitste config:
https://i.imgur.com/D1Rgnrr.png

dennissiemensma, 2018-03-05:
@Ernst79 bedankt voor je toevoeging, dan zal ik die ondersteuning er nog aan toevoegen.

@f0nt4 bedankt voor je toelichting, ik dacht dat je het over het tarief qua prijzen had, maar daar heb je dus uberhaupt geen behoefte aan? 
Het tarief qua hoog/laag is een gegeven die ik waarschijnlijk wel kan toevoegen, maar dan denk ik dat een los ticket daar een betere plek voor is. Het huidige ticket is echt van toepassing op het dashboard en daar passen deze gegevens minder goed bij. 
Ik kan dan de gegevens die in de applicatie staan onder `/admin/dsmr_datalogger/meterstatistics/` wellicht los beschikbaar maken via MQTT (kijk maar even of dat goed bijgewerkt wordt bij jou).

Ernst79, 2018-03-05:
Jij ook bedankt voor al het werk dat je er in steekt en hebt gestoken!

f0nt4, 2018-03-05:
@dennissiemensma Nieuw ticket is aangemaakt.
De tarieven (prijzen) zijn sowieso handig!

dennissiemensma, 2018-03-11:
Ik heb inmiddels een nieuwe opzet klaar staan. De dagwaarden van de huidige dag (zoals in dashboard) zijn daarin zowel via JSON in een enkel topic, of ruw via losse topics te ontvangen op MQTT.

Is er hier iemand geinteresseerd om dit uit te proberen om feedback te geven of dit afdoende is?

Uitproberen kan met:
```
sudo su - dsmr
git fetch
git checkout -b 367-mqtt-dashboard-stats origin/367-mqtt-dashboard-stats
./deploy.sh
```

Ernst79, 2018-03-11:
Heb het uitgeprobeerd en het werkt prima zo. Ik ontvang nu het dagtotaal via Mqtt en kan het uitlezen in bijv. Home Assistant. Ik heb het uitgesplitste topic getest. 

Mijn oude Mqtt gegevens komen ook nog steeds door. 

Hoe kom ik straks weer op de normale versie van dsmr-reader?

dennissiemensma, 2018-03-11:
Bedankt voor het uitproberen! Ik zal dit samenvoegen en dan zometeen releasen.

dennissiemensma, 2018-03-11:
Je kunt terug naar de normale versie met:
```
sudo su - dsmr
git checkout master
./deploy.sh
```

closed 2018-03-11 by dennissiemensma

dennissiemensma, 2018-03-11:
Uitgebracht in de [nieuwe release](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.14.0).

joopy, 2018-03-12:
Krijg het helaas niet werkend? MQTT: Broker/verbinding-configuratie ingevuld.

MQTT: Telegram (als onbewerkte string) configuratie aangezet maar komt niet binnen op de MQTT broker van home assistant. Mis ik iets?

f0nt4, 2018-03-12:
@joopy Heb je met een programma zoals mqtt.fx gekeken? Zo kan je snel controleren of dsmr reader verbinding maakt met de broker. Zelf maak ik gebruik van de gesplitste topics waarbij ik in eerste instantie in mqtt.fx een subscribe op de topics gedaan heb om te kijken wat er precies binnen komt.
Daarna in HA de topics gebruikt en werkt perfect.

dennissiemensma, 2018-03-12:
Joopy, gaat je vraag over MQTT in het algemeen of specifiek over de feature van dit ticket? 

Als het goed is zijn er 5 soorten MQTT-configuraties die je kunt aanpassen en de zesde (tweede hieronder) is de algemene broker die de daadwerkelijke verbinding (naar je HA?) maakt. 

[image]

joopy, 2018-03-14:
Thnx heb het al werkend gekregen. Home assistant onboard MQTT server laat zelfs met debug aan niet alles zien in de log's. Nu met mqtt.fx gekeken fijn programma btw. Het werkt!

FYI (dsmr/day-totals)
{"energy_supplier_price_electricity_delivered_2": "0.20255", "electricity2_cost": "2.09", "electricity2": "10.297", "electricity1_cost": "0.48", "gas": "3.195", "energy_supplier_price_electricity_returned_2": "0.00000", "gas_cost": "2.13", "electricity1_returned": "0.000", "energy_supplier_price_gas": "0.66458", "electricity2_returned": "0.000", "electricity_cost_merged": "2.57", "energy_supplier_price_electricity_delivered_1": "0.18440", "total_cost": "4.70", "electricity_returned_merged": "0.000", "electricity1": "2.601", "energy_supplier_price_electricity_returned_1": "0.00000", "electricity_merged": "12.898"}

Home assistant:
####DSMR

  - platform: mqtt
    name: DSMR dag Energy
    unit_of_measurement: "kWh"
    state_topic: "dsmr/day-totals"
    value_template: "{{value_json.electricity_merged }}"

f0nt4, 2018-03-14:
@joopy Mooi om te zien dat je het werkend hebt gekregen.
Ook hartelijk dank voor het delen van de HA config. Die kan ik wel weer gebruiken ipv dat ik nu allemaal losse telegrams verstuur.