dsmrreader/dsmr-reader #368: Gebruik van os.environ.get

ju5t, 2017-10-21:
Is het een idee om de instellingen van o.a. PostgreSQL in te kunnen stellen met os.environ.get? Dat maakt het configureren van Docker containers gemakkelijker.

Bijvoorbeeld in [postgresql.py](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/provisioning/django/postgresql.py) kunnen we `os.environ.get('DB_NAME', 'dsmrreader')` gebruiken om een default te houden en deze te overschrijven met een environment variabele in Docker.

Als het interessant is om dit hier op te pakken hoef ik de configuratie niet over te nemen in de bestaande Docker container. Dat heb ik nu als voorbeeld wel gedaan in [de fork van de Docker repo](https://github.com/ju5t/dsmr-reader-docker/blob/dockerfile-improvements/Docker/postgresql.py).

In de fork is het opgelost maar dat is niet echt DRY ;-)

dennissiemensma, 2017-10-21:
Bedankt voor je suggestie. Zolang het blijft werken voor omgevingen zonder de ENV, vind ik het een prima toevoeging.

ju5t, 2017-10-21:
Het blijft gewoon werken. De default staat in achter de environment variabele. Het kan voor iemand die niet zo bekend is met Python wel wat lastiger overkomen omdat er code om de instelling heen staat.

closed 2017-10-23 by dennissiemensma

dennissiemensma, 2017-11-24:
`v1.11` is [gereleased](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.11.0), inclusief dit ticket.