dsmrreader/dsmr-reader #370: Pvoutput upload zonder teruglevering

arvdsar, 2017-10-21:
Hi,
vandaag de DSMR Reader geinstalleerd en het werkt prima. Enige is dat het uploaden naar PVoutput niet lukt als er geen teruglevering wordt geregistreerd.

Ik heb wel een slimme meter maar geen zonnepanelen (meer). De upload naar PVoutput kreeg ik niet werkend totdat ik een handmatige aanpassing deed in /dsmr_pvoutput/services.py
onderstaande code heb ik uitgecommentarieerd.

`
if not capabilities['electricity_returned']:
     print(' - [!] PVOutput | No electricity return recorded by application!')
    return False
`

Kun je hier ook een optie van maken in de user interface? 

Groet,
Alexander

dennissiemensma, 2017-10-22:
Bedankt voor het melden, dit is een bug. In een eerdere iteratie voor de PVOuput-integratie was nog sprake van het uploaden van netto teruglevering. Dat is later gewijzigd naar alleen nette verbruik, dus deze check hoort er uberhaupt niet meer in.

dennissiemensma, 2017-10-23:
Ik heb het aangepast voor de volgende release.

closed 2017-10-23 by dennissiemensma

dennissiemensma, 2017-11-24:
`v1.11` is [gereleased](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.11.0), inclusief dit ticket.