dsmrreader/dsmr-reader #371: fonts via https laden

Ernst79, 2017-10-22:
Ik probeer DSMR reader te integreren in Home Assistant als een iPanel. Hiervoor dient de site volledig https te zijn, aangezien ik home assistant ook benader via https. 

Het blijkt echter dat als ik de DSMR pagina via https laad, dat er ook een viertal fonts via http worden gedownload. Chrome geeft hierover ook een waarschuwing middels een schildje rechts in de adresbar met als melding "this site is trying to load scripts from unauthenticated sources". Uit de developermode van chrome kan ik achterhalen dat dit 4 fonts zijn. 

> (index):1 Mixed Content: The page at 'https://www.mijnwebsite.nl:1234/' was loaded over HTTPS, but requested an insecure font 'http://fonts.gstatic.com/s/lato/v11/UyBMtLsHKBKXelqf4x7VRQ.woff2'. This request has been blocked; the content must be served over HTTPS.

> (index):1 Mixed Content: The page at 'https://www.mijnwebsite.nl:1234/' was loaded over HTTPS, but requested an insecure font 'http://fonts.gstatic.com/s/lato/v11/1YwB1sO8YE1Lyjf12WNiUA.woff2'. This request has been blocked; the content must be served over HTTPS.

> (index):1 Mixed Content: The page at 'https://www.mijnwebsite.nl:1234/' was loaded over HTTPS, but requested an insecure font 'http://fonts.gstatic.com/s/lato/v11/UyBMtLsHKBKXelqf4x7VRQ.woff2'. This request has been blocked; the content must be served over HTTPS.

> (index):1 Mixed Content: The page at 'https://www.mijnwebsite.nl:1234/' was loaded over HTTPS, but requested an insecure font 'http://fonts.gstatic.com/s/lato/v11/1YwB1sO8YE1Lyjf12WNiUA.woff2'. This request has been blocked; the content must be served over HTTPS.

Het moet volgens mij ook prima mogelijk zijn om deze fonts via https te laden, ik krijg althans als ik de links aanpas naar https ipv http gewoon de bestanden als download. Kan dit aangepast worden naar https?

dennissiemensma, 2017-10-23:
Bedankt voor je suggestie. Ik zal kijken of ik dat dat oplossen voor je.

dennissiemensma, 2017-10-23:
Ik heb ze aangepast voor de volgende release, zodat ze gewoon via dsmr-reader zelf geserveerd worden.

closed 2017-10-23 by dennissiemensma

Ernst79, 2017-10-23:
Dat is snel, bedankt. Ik wacht de release af.

dennissiemensma, 2017-11-24:
`v1.11` is [gereleased](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.11.0), inclusief dit ticket.