dsmrreader/dsmr-reader #372: Datalogger naar API

ju5t, 2017-10-24:
De huidige datalogger schrijft data direct weg naar de database. Ik wil de datalogger splitsen van de applicatie en dat betekent dat er geen directe toegang tot de database is. Dat kun je afvangen met de API maar dat betekent dat er code geschreven moet worden, buiten de applicatie om, die hetzelfde doet. Zonde als de functionaliteit al bestaat.

Ik zat zelf al te kijken of het misschien mogelijk is om bijvoorbeeld functies uit [`services.py`](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/services.py) te hergebruiken. Dat moet me wel lukken. Ik wil ook niet zomaar een feature request over de schutting kieperen.

Zijn er redenen, behalve gemak, waarom de datalogger nu via een Django model data weg schrijft?

dennissiemensma, 2017-10-24:
Dit project is een complete applicatie en is nooit opgezet om de datalogger los te gebruiken. 

Mocht je dat wel willen, zonder al te veel code te schrijven, dan raad ik je aan om [deze library te gebruiken](https://github.com/ndokter/dsmr_parser). Je hebt tevens ook véél minder requirements daarvoor nodig.

dennissiemensma, 2017-11-13:
Mocht je nog hulp nodig hebben, dan hoor ik het graag.

closed 2017-11-13 by dennissiemensma