dsmrreader/dsmr-reader #373: geen webinterface na installatie

jwpv, 2017-10-24:
Wat een prachtig project, het ziet er mooi uit en grandioos gedocumenteerd!
Maar het laatste stukje werkt niet, helaas.

Na de hele installatie krijg ik in mijn browser '504 Gateway Time-out', 'nginx/1.6.2'.
Bij supervisorctl status
dsmr_backend                     RUNNING    pid 1051, uptime 0:01:08
dsmr_datalogger                  RUNNING    pid 1037, uptime 0:01:09
dsmr_webinterface                RUNNING    pid 1029, uptime 0:01:09

In de logfile van de webinterface staat
[2017-10-24 22:00:42 +0200] [1310] [INFO] Booting worker with pid: 1310
[2017-10-24 22:01:43 +0200] [1029] [CRITICAL] WORKER TIMEOUT (pid:1310)
[2017-10-24 22:01:43 +0200] [1310] [INFO] Worker exiting (pid: 1310)

rebooten, update/upgrade helpt niet.
iemand enig idee waar ik het zoeken moet?

dennissiemensma, 2017-10-24:
Bedankt voor je melding. Op wat voor hardware draai je het project?

Daarnaast zou je kunnen kijken of #351, #323 of #252 je op weg kunnen helpen (kan andere oorzaak zijn). 

Daarnaast staan er mogelijk logfiles in de 'logs' map binnen het project.

jwpv, 2017-10-24:
Ik probeer de DSMR-reader te draaien op een raspberry pi 2b met raspbian jessie. Bij het installeren leek het meeste wel goed te gaan, behalve bij stap 7

  Running setup.py install for itypes

*** Error compiling '/tmp/pip-build-ar62g89v/jinja2/jinja2/asyncfilters.py'...
  File "/tmp/pip-build-ar62g89v/jinja2/jinja2/asyncfilters.py", line 7
    async def auto_to_seq(value):
            ^
SyntaxError: invalid syntax

*** Error compiling '/tmp/pip-build-ar62g89v/jinja2/jinja2/asyncsupport.py'...
  File "/tmp/pip-build-ar62g89v/jinja2/jinja2/asyncsupport.py", line 22
    async def concat_async(async_gen):
            ^
SyntaxError: invalid syntax

Gaan idee of dat er iets mee te maken heeft.

Tot mijn verbazing zijn de logfiles in het project leeg (0 bytes). 
Ik heb de issue's bekeken, maar die lijken ergens anders over te gaan.

Ik heb niet heel veel ervaring in linux, maar ik hoop toch dit project draaiende te krijgen. 
Erg veel dank voor het meedenken!

jwpv, 2017-10-25:
Na er een nachtje over geslapen te hebben werkt alles! Niet extra's gedaan, geen reboot oid. Kennelijk heeft het langer dan +-1 uur nodig om op te starten, na installatie (????)
Dank voor de hulp.

closed 2017-10-25 by jwpv

dennissiemensma, 2017-10-25:
Hmmm vreemd, meestal geeft de timeout aan dat het te lang duurt om iets te verwerken, vandaar mijn vraag om de hardware. Doet de applicatie het nu verder wel direct na een reboot?

Verder mooi dat het lijkt te werken overigens.

jwpv, 2017-11-15:
Volgens mij lukt het niet met een raspberri pi2b in combinatie met een DSMR meter 5.0 (iedere 10 seconden een telegram). De DSMR-reader website is bij herhaling niet bereikbaar. Pas als ik de datalogger uitzet lukt het weer na enige tijd. De load average was +-2. 
Ik heb de DSMR reader gemigreerd naar een pi3 en gebruik de pi2 nu alleen om de telegrammen door te sturen.

dennissiemensma, 2017-11-15:
Bedankt voor je toevoeging. Ik ben toevallig van de zomer ook overgestapt naar een Pi 3, maar dat was meer omdat ik dat ding al heel lang had liggen. :]

Ik zal dan de docs aanpassen dat een Pi 3 minimaal aanbevolen wordt.