dsmrreader/dsmr-reader #374: Problemen na installatie

lwolfs, 2017-10-25:
Hi,

Ik heb netjes de installatiehandleiding gevolgd (btw: prima werk geleverd), maar nu loop ik tegen een probleem op.

Platvorm: Raspberry 2B met debian 8.0
Domoticz draait ook op de raspberry.....

Symptoom: **Kan niet inloggen**

Tijdens installatie de volgende meldingen:
*** Error compiling '/tmp/pip-build-72bz686u/jinja2/jinja2/asyncfilters.py'...
  File "/tmp/pip-build-72bz686u/jinja2/jinja2/asyncfilters.py", line 7
    async def auto_to_seq(value):
            ^
SyntaxError: invalid syntax

*** Error compiling '/tmp/pip-build-72bz686u/jinja2/jinja2/asyncsupport.py'...
  File "/tmp/pip-build-72bz686u/jinja2/jinja2/asyncsupport.py", line 22
    async def concat_async(async_gen):
            ^
SyntaxError: invalid syntax

Geen idee of dit kwaad kan...

In /var/log/supervisor file dsmr_datalogger-stderr:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/__init__.py", line 364, in execute_from_comm
and_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/__init__.py", line 356, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 61, in handle
    self.run(**options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 23, in run
    telegram = dsmr_datalogger.services.read_telegram()
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 73, in read_telegram
    data = serial_handle.readline()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple acc
ess on port?)

In supervisor.log:
2017-10-25 10:56:09,945 INFO supervisord started with pid 743
2017-10-25 10:56:10,958 INFO spawned: 'dsmr_webinterface' with pid 746
2017-10-25 10:56:10,971 INFO spawned: 'dsmr_datalogger' with pid 751
2017-10-25 10:56:10,983 INFO spawned: 'dsmr_backend' with pid 752
2017-10-25 10:56:11,988 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2017-10-25 10:56:11,989 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2017-10-25 10:56:14,263 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2017-10-25 10:56:41,150 INFO exited: dsmr_datalogger (exit status 1; not expected)
2017-10-25 10:56:41,776 INFO spawned: 'dsmr_datalogger' with pid 819
2017-10-25 10:56:42,895 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

Ik heb PERSISTENT USB Binding aangezet, aangezien er ook een RFXCOM aan de Raspberry hangt.
De poort kan worden uitgelezen middels cu -l /dev/ttyUSB_P1 -s 115200 --parity=none -E q

Ik weet niet hoe ik nu verder moet gaan.....

jwpv, 2017-10-25:
Ik had precies hetzelfde: 
[https://github.com/dennissiemensma/dsmr-reader/issues/373](url)
Na een nachtje slapen werkt alles.

lwolfs, 2017-10-25:
@jwpv Krijg een 404 bij je URL... Ik ben erg nieuwsgierig

jwpv, 2017-10-25:
Nu werkt de link.

lwolfs, 2017-10-25:
Nog wat getest, gereboot.....niets werkt.
Als ik de URL gebruik, krijg ik constant 
**403 Forbidden

nginx/1.6.2**

In /var/log/supervisor blijven de stderr files leeg, en in ~/dsmr-reader/logs is alleen .dummy 4 bytes groot, rest is 0 bytes.

Processen draaien:
pi@raspberrypi:~ $ sudo supervisorctl
dsmr_backend                     RUNNING    pid 740, uptime 0:07:51
dsmr_datalogger                  RUNNING    pid 738, uptime 0:07:51
dsmr_webinterface                RUNNING    pid 737, uptime 0:07:51

(Ben ongeduldig....wil graag gegevens gaan inzien)

lwolfs, 2017-10-25:
Probleem opgelost:
Ik had de key uit settings.py niet opgeslagen als htpasswd
In /var/log/nginx/error.log stond dat de file werd geprobeerd te openen, en hij(zij?) bestond niet.

dennissiemensma, 2017-10-25:
Bedankt voor je melding.

* Ik heb de fout `serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?` nog niet eerder gezien. Werkt de applicatie verder wel qua data uitlezen uit de meter?

* Ik weet niet precies wat je bedoelt met de key uit `settings.py` en `htpasswd`, dat zijn twee losstaande dingen. Kan het zijn dat je htpasswd ingeschakeld hebt zonder een wachtwoord te genereren? Dat kan *elk wachtwoord naar eigen keuze* zijn overigens.

lwolfs, 2017-10-25:
De applicatie lijkt naar behoren te werken. Geen vreemde meldingen gezien.

Ik had idd bij de autorisaties de hekjes verwijderd en de key niet geplaatst in htpasswd.
Hekjes teruggezet, en file aangemaakt, en nu geen problemen meer :-)

Klein vraagje:
Ik zie in het dashboard staan: (Je kunt tot een week terugscrollen) 
Is detailinformatie na 1 week weg uit de database ?

dennissiemensma, 2017-10-25:
Momenteel wordt alle informatie nog bewaard, echter (met name voor DSMR v5 meters) zijn er op den duur zoveel metingen opgeslagen, dat de performance er onder kan lijden. Dus uiteindelijk komt hier nog (optionele) retentie voor via #72.

Alle informatie staat opgeslagen in de database. Zorg ervoor dat je regelmatig backup's maakt, helemaal wanneer je (standaard) een SD-kaart gebruikt.

closed 2017-10-25 by dennissiemensma