dsmrreader/dsmr-reader #375: Importeren data

jwpv, 2017-10-25:
Beste Dennis,
het is me gelukt DSMR-reader te installeren. Prachtig.
Nu wil ik graag oude data importeren.
Ik heb een jaar data in een mysql database en als het even kan wil ik ook data van oude facturen importeren zodat ik een overzicht krijg van meerdere jaren.
Ik heb zitten kijken in de postgresql database en als ik het goed zie gaat het om drie tabellen:

dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption
dsmr_datalogger_dsmrreading

Als ik de dsmr reader stop kan ik dan met een python script de data uit mysql halen en rechtstreeks in de postgresql database tabellen schrijven en dan daarna de geaggregeerde data in dsmr reader wissen en weer laten opbouwen?

Ik vond ook nog een methode om via de api oude data toe te voegen [https://balk.tweakblogs.net/blog/14421/script-om-p1-metingen-in-mysql-database-naar-dsmr-reader-api-te-sturen](url) 
maar dat lijkt me toch wat omslachtig. Bovendien heb ik de telegrammen niet.

De vraag is dus of het zou kunnen werken en of ik nog met wat anders rekening moet houden.

dennissiemensma, 2017-10-25:
Dat artikel is geschreven vóórdat de `v2` API in DSMR-reader zat (#230). Deze is een aantal maanden later gereleased. 

Het voordeel daarvan is dat je **geen telegram** meer nodig hebt en gewoon via REST de gegevens kan toevoegen.

Voor meer info, zie specifiek #226, wat volgende mij over dezelfde persoon/blog gaat. De API-docs staan hier: http://dsmr-reader.readthedocs.io/nl/latest/api.html#id2

dennissiemensma, 2017-10-25:
En ja, je kunt natuurlijk zelf ook proberen om directe queries te doen, maar ik vermoed dat de API wat simpeler is. Ook omdat er al een voorbeeld in de docs staat.

jwpv, 2017-10-25:
De API ziet er prachtig uit, en duidelijke voorbeelden. Dat moet lukken. Het voelt wel vreemd om data te sturen naar een url op dezelfde Pi. Zal dat niet een eeuwigheid gaan duren? Maar ja, in principe eenmalig.

Als ik het toch direct doe, is het dan voldoende om de tabel dsmr_datalogger_dsmrreading te vullen en `./manage.py dsmr_backend_delete_aggregated_data` te doen?

dennissiemensma, 2017-10-25:
Ja klopt. Iemand had daar overigens eerder wel een issue mee en [dat op deze manier opgelost](https://github.com/dennissiemensma/dsmr-reader/issues/328#issuecomment-307013093), mocht je tegen hetzelfde aanlopen.

In principe vult die API-call inderdaad `dsmr_datalogger_dsmrreading`.

dennissiemensma, 2017-11-13:
Mocht je nog hulp nodig hebben, dan hoor ik het graag.

closed 2017-11-13 by dennissiemensma