dsmrreader/dsmr-reader #376: Nieuwe meter

TopdRob, 2017-10-25:
Hoi Dennis,

Mijn slimme meter is vervangen voor een andere en die begint nu weer op 0. Is het mogelijk om de standen van de nieuwe meter op te hogen met de standen van de oude meter? Ik heb nu een negatief verbruik?

Bedankt,
Robert

dennissiemensma, 2017-10-26:
Zie #242. Ik heb hier momenteel nog geen oplossing voor.

Als het goed is krijg je de dag erna wél de goede standen (hij kijkt telkens naar de voorgaande dag). Vervolgens zou je in de admin-interface eenmalig handmatig de foutieve dag kunnen corrigeren (zul je zelf even moeten rekenen).

dennissiemensma, 2017-11-04:
@TopdRob is bovenstaande gelukt?

TopdRob, 2017-11-04:
Hoi @dennissiemensma,

Bedankt voor je hulp. Ik heb de dag op 0 gezet en daarmee is het opgelost!

dennissiemensma, 2017-11-05:
Top!

closed 2017-11-05 by dennissiemensma