dsmrreader/dsmr-reader #377: DSMRREADER en DOMOTICZ samen

lwolfs, 2017-10-28:
Hi,

Ik heb ' last'  van het volgende fenomeen:
Op mn RASPBERRY PI staat Domoticz en DSMRREADER geinstalleerd. Werkt prima samen, alleen krijg ik nu in Domoticz de melding "Error: Serial Port closed!... Error: End of file" en collecteert hij geen gegevens meer. DSMRREADER werkt als een tierelier.
Aangezien beide applicaties ' zuigen'  aan de P1 kabel, zou hier wel eens een conflicht kunnen ontstaan als ze beiden tegelijkertijd 'zuigen' .
Lijkt me handiger dat eentje ' zuigt'  en de ander gebruik maakt van die gegevens.
Misschien vertel ik nu iets dat onzin is, of allang is uitgewerkt, maar ik weet het nu niet, en ben op zoek naar een oplossing.

Misschien komt iemand met de vraag waarom EN DSMRREADER EN Domoticz ?
Mijn zonnepanelen worden in Domoticz gepolled (via SolarEdge API) en verwerkt, en kan ik zowel stroomverbuik/levering en gegevens zonnepanelen naast elkaar zetten. Alleen Domoticz is erg beperkt qua grafieken, en ziet DSMRREADER er beter uit (persoonlijke mening!).

BTW: zowel Domoticz database als DSMRREADER database op een aparte partitie van een goede USB stick gezet. Nee, niet 1 maar 2 partities.

dennissiemensma, 2017-10-28:
Bedankt voor je ticket, wat is concreet je vraag? Hoe je ze beide tegelijkertijd van de kabel gebruik kan laten maken?

dennissiemensma, 2017-10-28:
Ik dacht trouwens dat een [telefoonsplitter](https://github.com/dennissiemensma/dsmr-reader/issues/187#issuecomment-270456785) ook al uitkomst kan bieden. En ik kan me herinneren dat er softwarematig ook een oplossing voor was, maar ik kan daar zelf niet zo 1-2-3 op komen.

lwolfs, 2017-10-28:
DSMRREADER en DOMOTICZ zijn beiden geïnstalleerd op dezelfde raspberry; een splitter is niet nodig.
Maar het feit dat beide applicaties data ophalen van de slimme meter, kan, misschien, een probleem opleveren, als ze dit gelijktijdig doen, dus beiden de USB poort openen.
Mijn vraag is of dit een probleem zou kunnen zijn, en of bv mogelijk is om Domoticz de data te laten lezen, en DSMREADER de data van domoticz te laten krijgen of omgekeerd..

dennissiemensma, 2017-10-28:
Ik kan daar helaas geen antwoord op geven. Zelf gebruik ik slechts één applicatie op één poort. Ik denk niet dat het mogelijk is om de applicaties data van elkaar te laten lezen. 

Je zult dan een software-matige oplossing moeten vinden voor het clonen van de poort op je raspberry. Zoals aangeven is dat volgens mij wel mogelijk (wellicht via `socat`), maar ik heb daar persoonlijk geen ervaringen mee. :]

Zolang ze beiden blijven werken zal het geen probleem zijn.

lwolfs, 2017-10-28:
Thx. Ik ga eens stoeien. Laat dit topic nog even open.
Als er iets te melden is, zal ik het hier plaatsen.

lwolfs, 2017-10-30:
Probleem binnen domoticz, denk ik. Ik sluit de case bij deze. Thx voor het meedenken Dennis !

closed 2017-10-30 by lwolfs

glsf91, 2018-11-27:
Ik heb het dupliceren van een seriele poort op Ubuntu aan de praat met:
#!/bin/bash
cd /home/john/serial_splitter

echo "Set speed ttyUSB1"
stty -F /dev/ttyUSB1 115200 clocal ignpar -ixon -opost -isig -icanon -iexten -echo -icrnl

echo "Create fifo1 and fifo2"
mkfifo fifo1 fifo2
echo "Start tee"
tee </dev/ttyUSB1 fifo1 >fifo2 &
echo "Create /dev/ttyUSB1a"
sudo socat -u GOPEN:fifo1 pty,link=/dev/ttyUSB1a,raw,echo=0,group=dialout,mode=0660 &
echo "Create /dev/ttyUSB1b"
sudo socat -u GOPEN:fifo2 pty,link=/dev/ttyUSB1b,raw,echo=0,group=dialout,mode=0660 &

Ik heb DSMR-reader dan op /dev/ttyUSB1b en domoticz op /dev/ttyUSB1a.
Kabel zit op /dev/ttyUSB1