dsmrreader/dsmr-reader #378: Processing of telegrams stalled

allureGOT, 2017-10-30:
Hi!

Ik heb DSMR (ver. 1.10) draaien op een Intel NUC, OS: Ubuntu 16.04.
Nu is het verwerken van de telegrammen gestopt, als ik op de frontpage van DSMR kijk lijkt deze te zijn gestopt om 3 uur snachts, dit heeft waarschijnlijk te maken met de zomer/wintertijd.

Hoe kan ik dit proces weer op gang helpen?
Ik heb de 3 systemctl processen al handmatig gestopt en gestart en ook een reboot van het hele system helpt niet.

https://ic.tweakimg.net/camo/61f3f2bac747dbc7784aabb715f526274fdc7bb8/?url=http%3A%2F%2Fi65.tinypic.com%2F2rwkl6f.png

dennissiemensma, 2017-10-30:
Bedankt voor je melding. Heb je al in de logs gekeken? Zie deze pagina voor meer info: http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

allureGOT, 2017-10-30:
Is de logfile dsmreader.log staan met name CRC mismatches en rejected telegrams.
Wat kan hier de oorzaak van zijn? een verschil in tijd?

dennissiemensma, 2017-10-30:
Dat klopt wel, standaard logt die alleen fouten (en die komen soms voor). 

Echter, ik denk dat je moet zoeken in de log van `dsmr_backend`, gezien die de verwerking doet. Staat daar wat in? Logfile staat ergens in `/var/log/supervisor/`.

allureGOT, 2017-10-30:
Er staan 6 logfiles met de naam dsmr_backend in en deze lijken me geen foutmeldingen te bevatten:

http://tinypic.com/r/jrqr2u/9

dennissiemensma, 2017-10-30:
Het lijkt erop dat er iets met de databaseverbinding of een query mislukt is. Heb je toevallig de complete stacktrace van die screenshot?

Je kunt overigens hier op Github gewoon bijlages uploaden door ze te slepen of door op de link onder het tekstveld te klikken.

dennissiemensma, 2017-10-30:
Wat ook makkelijk is is om een live tail te down op de backend: `sudo supervisorctl tail -f dsmr_backend`

allureGOT, 2017-10-31:
Misschien dat je wat bijzonders kan zien in de logfile?
dsmr_backend.log](https://github.com/dennissiemensma/dsmr-reader/files/1429949/dsmr_backend.log)

dennissiemensma, 2017-10-31:
Ik denk dat je mag zoeken naar het ID en die vervolgens verwijderen.

* `sudo su - postgres`
* `psql dsmrreader`
* `select * from dsmr_datalogger_dsmrreading where processed is null order by timestamp asc limit 5;`
* Als het goed is krijg je nu een overzicht van 5 onverwerkte metingen, waarbij de bovenste hetzelfde datum/tijdstip weergeeft als in de foutmelding.
* Zoek het ID vooraan.
* Verwijder hem met: `DELETE from dsmr_datalogger_dsmrreading where id = XXX;` (XXX = het ID)

Als het goed is gaat die vervolgens verder.

allureGOT, 2017-10-31:
In die tabel lijken 0 rows te zitten die aan dit criterium voldoen. :)
[image]

dennissiemensma, 2017-10-31:
Ik ben niet wakker, het gaat om een andere tabel, namelijk degene in je foutmelding.

* `select * from dsmr_consumption_electricityconsumption order by read_at desc limit 5;`
* `DELETE from dsmr_consumption_electricityconsumption where id = XXX`;

dennissiemensma, 2017-10-31:
Mocht dat het niet oplossen, dan zou je kunnen kijken naar de betreffende meting waar het over gaat, die staat wel in de eerder genoemde tabel, maar de query is dan iets anders.

* `select * from dsmr_datalogger_dsmrreading where processed is false order by timestamp asc limit 5;`

`dsmr_datalogger_dsmrreading` is de brontabel, waar alle metingen in staan. `dsmr_consumption_electricityconsumption` is waar alle metingen naar aggregeerd worden.

allureGOT, 2017-10-31:
De 01:00:00 tijden bestaan inmiddels niet meer.

Kan het probleem niet eerder in het feit zitten dat de waarde in tabel "electricity_delivered_1" al bestaat? (12986)

[image]

allureGOT, 2017-10-31:
Ik denk erover om de bestaande DB te droppen en gewoon opnieuw te beginnen.
In de logs komt nog steeds die "01:00:00" terug maar die bestaan niet meer.

dennissiemensma, 2017-10-31:
Top, bedankt voor je update. De foutmelding ging over het `read_at` veld, dus het kon niet te maken hebben met `electricity_delivered_1`. Blijft overigens raar dat dit heeft kunnen gebeuren, maar het inderdaad niet de bedoeling dat hetzelfde tijdstip (in dezelfde tijdzone) voorkomt. Laat maar weten of dit het helemaal oplost.

allureGOT, 2017-10-31:
Ok, ik heb nog 5 entry's moeten verwijderen, maar alle telegrammen zijn verwerkt.
Ik laat m nu even lopen en zal morgen melden of het probleem opgelost is om weer terug gekomen is.

allureGOT, 2017-11-01:
Het  probleem heeft zich vandaag niet meer laten zien.

dennissiemensma, 2017-11-01:
Top, goed om te horen!

allureGOT, 2017-11-07:
Ik weet hoe t euvel te verhelpen, de verwerking van de telegrammen loopt echter wel elke dag een keer vast.

dennissiemensma, 2017-11-07:
Met wat voor fout loopt die vast, dezelfde als hierboven?

allureGOT, 2017-11-08:
Het is weer zover:
[image]

Verwijderen van ID 1272130 zorgt ervoor dat de verwerking weer doorgaat:
[image]

dennissiemensma, 2017-11-08:
Hmmm, vanzelfsprekend hoort dit niet te gebeuren. Je bent overigens de enige die hier last van lijkt te hebben. 
Mogelijk dat ik later deze week nog kan kijken naar een oplossing zodat die dubbele waarden gewoon automatisch overslaat.

allureGOT, 2017-11-12:
Misschien is dat wel een dingetje voor in de volgende release 👍

dennissiemensma, 2017-11-13:
Ik heb nog even voor je gekeken. Kan het zijn dat je tevens `Groeperen per meting` hebt aanstaan?

dennissiemensma, 2017-11-13:
Ik heb een mogelijke fix gemaakt. Je kunt het vrij makkelijk uitproberen met:

* `sudo su - dsmr`
* `git fetch`
* `git checkout -b 378-duplicate-timestamp-handling origin/378-duplicate-timestamp-handling`
* `./deploy.sh`

Als het goed is herlaadt dat ook het proces met de fix in de code.

Je kunt weer terug met:
* `sudo su - dsmr`
* `git checkout master`
* `./deploy.sh`

allureGOT, 2017-11-14:
Ik zal vanmiddag deze fix op t systeem gaan uitrollen, ik laat je de uitkomst ASAP weten.

dennissiemensma, 2017-11-14:
Dat is een goeie, maar puur tekstueel. Het is een overblijfsel van de DSMR v4 ondersteuning, dus ik zal die tekst ook meteen even aanpassen. Als het goed is groepeert hij gewoon per meting.

Overigens mooi dat je de groepering op per meting stond, want dat was ook de enige manier waarop ik het zelf kon reproduceren. Ik had er ooit al een test voor geschreven, die deze situatie moest nabootsen, maar die bleek niet (meer) te kloppen. Vandaar dat het ook niet zovaak voorkomt (geen standaardwaarde). Zowel de test als de fix zouden moeten werken.

Je hoeft de groepering dus niet te wijzigen als je dat niet wilt. Het is/was waarschijnlijk dus alleen een issue in combinatie met de groepering per meting. Ik hoor graag of de branch het oplost.

allureGOT, 2017-11-14:
Ok, version 1.11b4 heb ik deployed, ik ga de groepering weer op 10 seconden zetten.

dennissiemensma, 2017-11-16:
Is het issue sindsdien nog voorgekomen?

allureGOT, 2017-11-16:
Tot nu toe heeft het probleem zich niet meer voorgedaan.

dennissiemensma, 2017-11-16:
Oke top, dan zal ik de wijziging opnemen voor de volgende release. Tot die tijd kun je deze code gerust draaien.

allureGOT, 2017-11-16:
Ok, ik houd t wel even nauwlettend in de gaten, zodra het misgaat meldt ik me.
Zoniet, zal ik bij de volgende release weer op de main-versie overstappen.

closed 2017-11-22 by dennissiemensma

dennissiemensma, 2017-11-24:
`v1.11` is [gereleased](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.11.0), inclusief dit ticket.