dsmrreader/dsmr-reader #381: 502 Bad Gateway

dommels, 2017-11-15:
Ik heb sinds kort problemen met de webservice. 
De backend en de datalogger service draaien, alleen de webinterface geeft een fatal error:
dsmr_webinterface                FATAL      Exited too quickly (process log may have details)

Uit de nginx error log haal ik dit:
2017/11/15 21:33:16 [crit] 3959#0: *44 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstream, client: 192.168.2.105, server: _, request: "GET / HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket:/", host: "192.168.2.3"

Enig idee?

dennissiemensma, 2017-11-16:
De webserver (Nginx) verbindt naar de applicatie (in Gunicorn). Kun je eens kijken welke foutmelding Gunicorn geeft? Of het process tailen in Supervisor?

Er staat meestal wel een foutmelding in waarom Gunicorn niet draait/kan starten.

dennissiemensma, 2017-11-16:
* `sudo supervisorctl tail -f dsmr_webinterface`
* `sudo view /var/log/supervisor/dsmr_webinterface.log`

HugoDaBosss, 2017-11-16:
Ik had hetzelfde probleem en ik heb het net opgelost door het volgende te doen. Dit is misschien niet de manier hoe het zou moeten, maar het werkt.

In /var/tmp/ zijn 2 bestanden van gunicorn, namelijk gunicorn--dsmr_webinterface.pid en gunicorn--dsmr_webinterface.socket. Deze 2 bestanden heb ik verwijderd en daarna mijn Raspberry Pi opnieuw opgestart. Alles werkt nu weer naar behoren.

dommels, 2017-11-16:
@dennissiemensma een proces tail levert niets op. 

In /var/tmp/ heb ik maar 3 bestanden staan:
dsmrreader--dsmr_backend.pid  
dsmrreader--dsmr_datalogger.pid  
gunicorn--dsmr_webinterface.pid

 gunicorn--dsmr_webinterface.socket mis ik blijkbaar? (ook na een reboot)

dennissiemensma, 2017-11-16:
Wat geeft de logfile aan? Het tweede commando hierboven dat ik neer had gezet.

dennissiemensma, 2017-11-16:
Eventueel kun je ook de webserver handmatig draaien, om te kijken of er een foutmelding komt (Supervisor voert simpelweg één command uit en herstart die als die stopt.)

```
cd /home/dsmr/dsmr-reader/
sudo -u dsmr /usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--TEST.socket --pid /var/tmp/gunicorn--TEST.pid dsmrreader.wsgi
```

Ik krijg dan na een paar seconden zoiets te zien:
```
[2017-11-16 20:51:41 +0000] [32113] [INFO] Starting gunicorn 19.7.1
[2017-11-16 20:51:41 +0000] [32113] [INFO] Listening at: unix:/var/tmp/gunicorn--TEST.socket (32113)
[2017-11-16 20:51:41 +0000] [32113] [INFO] Using worker: sync
[2017-11-16 20:51:41 +0000] [32116] [INFO] Booting worker with pid: 32116
```

HugoDaBosss, 2017-11-16:
Bij mij stonden de volgende regels in de logfile

[2017-11-16 20:10:57 +0000] [1339] [INFO] Starting gunicorn 19.7.1                 Error: Already running on PID 844 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)                                                                   [2017-11-16 20:11:01 +0000] [1342] [INFO] Starting gunicorn 19.7.1                 Error: Already running on PID 844 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)                                                                   [2017-11-16 20:09:27 +0000] [1265] [INFO] Starting gunicorn 19.7.1                 Error: Already running on PID 844 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)                                                                   [2017-11-16 20:09:32 +0000] [1270] [INFO] Starting gunicorn 19.7.1                 Error: Already running on PID 844 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)                                                                   [2017-11-16 20:09:38 +0000] [1276] [INFO] Starting gunicorn 19.7.1                 Error: Already running on PID 844 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

dennissiemensma, 2017-11-16:
Ik denk dat je al die PID-files kan verwijderen en vervolgens `sudo supervisorctl restart all` kan doen. Het lijkt een beetje op wat HugoDaBosss beschrijft. Wellicht lost dat het inderdaad op, ik zie geen andere fout staan.

HugoDaBosss, 2017-11-16:
[image]

Dit vond ik ook eigenaardig, na een restart van mijn Raspberry Pi startte de interface in eerste instantie op, maar toch stopte hij achteraf weer.

dennissiemensma, 2017-11-16:
Hmm vreemd. Kreeg je ook geen foutmelding toen je Gunicorn handmatig draaide?

HugoDaBosss, 2017-11-16:
Ik heb toen niet geprobeerd Gunicorn handmatige te draaien, dus geen idee of dat zou hebben gewerkt.

Ik heb toen wel de django development server gestart met ./manage.py runserver, en daarbij gaf hij toen geen errors.

dennissiemensma, 2017-11-16:
De runserver roept onderwater hele andere code aan, vooral wanneer Gunicorn het probleem is. Maar als je het niet erg vind om een oudere versie van Gunicorn te draaien, dan prima. Je zult alleen wel bij elke release/update weer even moeten downgraden. :]

dennissiemensma, 2017-11-16:
Ooh ik lees telkens de reacties bij een ander persoon. In het geval van HugoDaBosss was het weghalen van de PID-files de enige oplossing. 
Voor dommels is downgraden voor nu prima, tenzij je nog verder wilt debuggen.

closed 2017-11-16 by dennissiemensma

dommels, 2017-11-16:
Ik hou er wel rekening mee in de toekomst. voor nu werkt het weer, bedankt.