dsmrreader/dsmr-reader #383: Total NOOB question about REST API

casperleenheer, 2017-11-18:
Hi Dennis,

I'm a total NOOB around REST and the API calls. I want to get some measurements from dsmr-reader by sending an url. I only don't know how to add the authentication (X_HEADER) to the url. So basically one url in my browser should give me the result. By the way, very nice product.

Thanks,

Casper

dennissiemensma, 2017-11-18:
Ik denk niet dat je het via de browser kan doen, maar je kunt wel een plugin/app gebruiken zoals `Postman`. 
Daarin kun je namelijk naast een URL, ook headers opgeven. Er is tevens ondersteuning voor het uitlijnen van de API response (JSON).

[image]

casperleenheer, 2017-11-18:
Dank je wel voor het antwoord. Ik heb het alsnog op een andere manier opgelost.

closed 2017-11-18 by casperleenheer